EdgeTX/edgetx #1347: [Feature request] Full duplex modules shouldnt need to skip RC packets...

JyeSmith, 2022-01-05:
At the moment Lua/tlm packets are sent in place of RC packets as required.  This shouldn't be needed with full duplex due to there being no chance of collision, unlike half duplex on the sport pin.  Along with the increased baud rates there is plenty of time to send these packets between RC packets.

raphaelcoeffic, 2022-01-05:
ATM, the LUA packets are sent asynchronously by copying into an output buffer which is picked up when sending the RC packets:
```C++
static void setupPulsesCrossfire(uint8_t idx, CrossfirePulsesData* p_data,
                                 uint8_t endpoint)
{
#if defined(LUA)
  if (outputTelemetryBuffer.destination == endpoint) {
    memcpy(p_data->pulses, outputTelemetryBuffer.data,
           outputTelemetryBuffer.size);
    p_data->length = outputTelemetryBuffer.size;
    outputTelemetryBuffer.reset();
  } else
#endif
  {
    if (moduleState[idx].counter == CRSF_FRAME_MODELID) {
      p_data->length = createCrossfireModelIDFrame(p_data->pulses);
      moduleState[idx].counter = CRSF_FRAME_MODELID_SENT;
    } else {
      p_data->length = createCrossfireChannelsFrame(
          p_data->pulses,
          &channelOutputs[g_model.moduleData[idx].channelsStart]);
    }
  }
}
```

As the data itself is then sent via DMA, we'd need to wait for the first transfer (RC packets) to be finished before flushing the telemetry buffers.

On the other hand, sending directly from the LUA function has a certain probability of colliding as well, so that we would need some kind of synchronisation as well.

I think this needs a bit of thinking to be done properly.

raphaelcoeffic, 2022-01-05:
The other option would be to move this code to the part that polls the telemetry RX buffers.

Woschd, 2022-01-06:
Does #if defined(LUA) at the code shown above mean the compiler option, which was renamed shortly?
(End the LUA != LUA confusion #1282)
If so, I think it doesn't make too much sense to have sportTelemetryPush() only in combination with Lua_Mixer = Yes.

pfeerick, 2022-01-07:
No, that is un-related. LUA=Y/N still exists (i.e. the actual LUA on/off option), only the LUA=YES string state was broken out to it's own option.