EdgeTX/edgetx #1349: invalid node in radio.yml message after successfull conversion to Santa 2.6 RC2

ParkerEde, 2022-01-05:
I've convert the edgetx 2.5 SD Card with simulator succesfull. After restarting the simulator i got this message:
[image]

Is this a generally issue or only my radio.yml file?

raphaelcoeffic, 2022-01-05:
We would need to see the files (binaries you converted from, and resulting YAML files). When you say you "convert[ed] the edgetx 2.5 SD Card with simulator succesfull[y]", do you mean in Companion? Or in the standalone simulation? How exactly did you proceed?

ParkerEde, 2022-01-05:
I mean the standalone simulator in
"C:\Program Files\EdgeTX\Companion 2.6\simulator.exe"
with DataSource "SD Path"
The SD Card Path contains the edgeTX 2.5 files (with bin files).
The first launch shows a message "SD card conversion required". This procedure ends with success. Then I close the simulator and start it again. Then the message with invalid node appears.
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/7814247/RADIO.zip)

PR #1350 (2022-01-07): YAML Trainer data: iterate through nodes rather than assuming their presence
Resolves #1349

Fixes reading this:
```yaml
trainer: 
   calib: 
      2:
         val: -512
      3:
         val: -3
   mix: 
      0:
         srcChn: 3
         mode: ADD
         studWeight: 100
      1:
         srcChn: 1
         mode: ADD
         studWeight: 100
      2:
         srcChn: 2
         mode: ADD
         studWeight: 100
      3:
         srcChn: 0
         mode: ADD
         studWeight: 100
```
commit eb90b4a6 (2022-01-07): YAML Trainer data: iterate through nodes rather than assuming their presence (#1350)

Resolves #1349
commit 2adca814 (2022-01-07): YAML Trainer data: iterate through nodes rather than assuming their presence (#1350)

Resolves #1349
closed 2022-01-07 by pfeerick