EdgeTX/edgetx #1351: Mainview 1-5 all options (Top bar, Flight mode...) are deactivated in 2.6 RC2

ParkerEde, 2022-01-05:
I open a otx file created in edgetx 2.5 where the Main view 1-5 options like Top bar, Flight mode, Sliders and so on was activated.
In edgetx 2.6 all these options are deactivated.
After conversion with the standalone simulator, these options are deactivated too.
It seems, that there is a migration 2.5 to 2.6 issue regarding the main view options.

dwheel, 2022-01-05:
Jumper T16, and Radiomaster Tx16 are in use here. I only have 1 view, but it does not load correctly in the companion simulator. It does load correctly in the transmitters.

I will attach the model and radio files:
[model2.zip](https://github.com/EdgeTX/edgetx/files/7816097/model2.zip)
[radio.zip](https://github.com/EdgeTX/edgetx/files/7816098/radio.zip)
.

koaxheli, 2022-01-05:
Transmitter Radiomaster TX16S
In the companion all option fields are returned with 0.
In the transmitter, it is OK.
ScreenData: 
   0:
      LayoutId: "Layout1x1"
      layoutData: 
         Zones: 
            0:
               WidgetName: "showal0.9"
               widgetData: 
                  Options: 
                     0:
                        type: Boolean
                        value: 
                           bool value: 1
                     1:
                        type: color
                        value: 
                           Color: 0xF8FCF8
                     2:
                        Type: color
                        Value: 
                           Color: 0x0000
         Options: 
            0:
               Type: Boolean
               value: 
                  bool value: 1
            1:
               type: Bool
               value: 
                  boolValue: 0
            2:
               type: Bool
               value: 
                  boolValue: 1
            3:
               type: Bool
               value: 
                  boolValue: 1
            4:
               type: Bool
               value: 
                  boolValue: 0
[image]
[image]
[image]
[image]

raphaelcoeffic, 2022-01-06:
From what I can tell, this seems windows specific. It could not be reproduced on Linux or Mac.

PR #1358 (2022-01-06): Fix bit mask generation
Avoids overflowing when the bit mask should be 32bits long.

Resolves #1351
commit d46067fa (2022-01-06): Fix bit mask generation (#1358)

Avoids overflowing when the bit mask should be 32bits long.

Resolves #1351
commit 234ebb26 (2022-01-07): Fix bit mask generation (#1358)

Avoids overflowing when the bit mask should be 32bits long.

Resolves #1351
closed 2022-01-06 by pfeerick