EdgeTX/edgetx #1354: Companion for Linux - some sub windows are frozen

jfrickmann, 2022-01-06:
I run Ubuntu Linux in WSL  under Win11.
I have tried with Companion and Simulator 26-27 built with Docker, and I have tried with the Downloaded AppImage from the latest Companion build on GitHub.

The following sub windows are frozen, i.e. they do not respond to any mouse events. They can only be shown and hidden again by pressing the F-keys.
- Radio Outputs (F2)
- Telemetry Simulator (F4)
- Trainer Simulator (F5)
- Debug Output (F6)

If someone with a native (Ubuntu) Linux machine could please test, to see if it is a Companion issue or an issue with Win11 WSL?

wimalopaan, 2022-01-06:
I use Companion26/27 and Simulator26/27 all time time under Linux (Arch Linux) and never experienced such problems.

pfeerick, 2022-01-06:
For Companion, I usually run the Linux version in a Ubuntu 20.04 VM when I need to test a Linux version as I don't have native linux running on my main box atm. Running the 2.6.0-RC2 appimage, I have no trouble fiddling with the trainer and telemetry windows... 

I also just tried the native Windows Companion for 2.6.0-RC2, and that was fine... is there any particular reason you need to run companion under WSL rather than a native windows version? Or is this because simulator seems to be bugged - or are you having problems with the native windows version, as there has been mention of issues (and of it working) on Windows 11. 

My primary OS atm is Windows 10, and I build using the edgetx docker image using WSL2, and use vcxsrv when I need to access a GUI like `simu`. I just tried the Companion appimage via that docker container + vcxsrv pipeline, and it seems to be fine there also - so unfortunately it seems like there must be something bugged about the Windows 11 + WSLg combination.

stronnag, 2022-01-06:
I can also confirm that Companion26 and Companion27 work correctly in Linux, both Arch (built locally) and Ubuntu 20.04 (official app-image). Used recently and extensively to test the inav lua widget for #1346.

richardclli, 2022-01-06:
I don't have a problem either.  I just downloaded the AppImage and run it in my OpenSUSE Leap 15.2 (with proper libraries packages installed)

elecpower, 2022-01-06:
Ubuntu 20.04 main branch compile and no issues with window focus.

jfrickmann, 2022-01-08:
Thanks all for the feedback! So I will conclude that this is an issue with the Wayland based X11 server in Win11 WSL and get a third party X11 server up and running.

closed 2022-01-08 by jfrickmann