EdgeTX/edgetx #1355: update the internal module with EdgeTX

KirkTz, 2022-01-06:
hi ,new user to Edgetx 2.5 (Tx16s) ,i need to update the internal module ,i just download the latest .. mm-stm-ppm-aetr-v1.3.3.0.bin but i 'm getting thiserror msg when i try to flash it,any ideas ?? what i'm missing ,,and also can i copy and run the lua scripts that was in the opentx scripts folder ?? 
[image]
?

pfeerick, 2022-01-06:
`mm-stm-ppm` is the *wrong* firmware for the internal MPM on the TX16S. Go back to the [MPM Firmware Download](https://downloads.multi-module.org/) page, make sure you pick the RadioMaster TX16S, and you'll get the correct firmware for your TX16S internal MPM. You can also then get the current/matching version of the Lua scripts. 

[image]

closed 2022-01-06 by pfeerick

KirkTz, 2022-01-06:
Thank you very much