EdgeTX/edgetx #1356: Provide a Quad/Drone specific Build of each release

tobiasbischoff, 2022-01-06:
The current build of EdgeTX has many features enabled that are irrelevant to drone pilots, but clutter the menu (HELI and FLIGHT MODES for instance). On the other hand I feel that 80% of the people using EdgeTX are pilots of multirotors that don't need these features enabled. I would propose providing an official "Drone" or FPV Build that removes these not needed features. That would make it way easier for Pro's to navigate the menus and will not confuse new user so much. 

**Describe alternatives you've considered**
I know that in principle everyone can just build his own Edition of EdgeTX but i don't consider this trivial. While not a developer i consider myself kinda confident with the Linux CLI and build tools and I was not able to do it starting off with a fresh cloud server and this howto

https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04 

**Additional context**
OpenTX had the ability to select build options in the configurator and download a customized build, some people (like me) might be used to this. If its easier, restoring this functionality would be sufficient as well but I feel a dedicated drone build would be better.

raphaelcoeffic, 2022-01-06:
Regarding the presence or absence of the menus, we will introduce shortly some settings to hide certain menus which are not comply used. That being said, the specific build options will be supported once the online build platform is up and running.

pfeerick, 2022-01-06:
Just out of curiosity, did you try out out the gitpod online development environment (badge/link is at the top of the readme)  - as that does not require any setup as it uses the docker build container and is pre-configured - you would be able to give your cmake build parameters, make the firmware, and then download it. 

However, that is just a stopgap, as the online build server is progressing, and once that is up, you'll be able to pick your build options like you used to be able to with with Companion. I will probably also do a no_heli variant when I do the builds for 2.6.0 (as I will be doing the various to do all the language builds), but probably won't go any further as what makes sense for drones is really up to interpretation as well - i.e. I use the flight modes with one of mine 😆

tobiasbischoff, 2022-01-06:
> Just out of curiosity, did you try out out the gitpod online development environment (badge/link is at the top of the readme) - as that does not require any setup as it uses the docker build container and is pre-configured - you would be able to give your cmake build parameters, make the firmware, and then download it. Probably needs advertising more as it is much easier to use for simple "just want to compile my own custom fw" scenarios.

Thanks for pointing that out. I did not really know what that is, but i literally clicked two times with my mouse and now have a custom build! Awesome. Thanks. Yes i guess that somehow needs better advertisement.

lshems, 2022-01-06:
I have introduced in my apps a basic and expert mode. The basic mode will hide some screens and options.

You can switch at anytime without losing the full capabilities.

:)

pfeerick, 2022-01-07:
Feel free to add a link ;)

gagarinlg, 2023-03-18:
Won't happen. But some tabs might be enabled/disabled per model in the future

closed 2023-03-18 by gagarinlg