EdgeTX/edgetx #1359: 2.6 RC2 + Windows Companion YAML saving wrong "LS" EDGE Values to all Models

Nerenye, 2022-01-06:
Occurs on Companion 2.6 Compiled on Jan 3. 22 
And
Horus X12s VERS: 2.6.0 (5b21bdbe)

To reproduce just load your models from TX into Companion and directly write it back to your TX without editing anything
Your Logic Switches with EDGE function now read like "Correct first Value":56.0

So Like:
EDGE L4 0.5:<< 
is now
EDGE L4 0.5:56.0

But its not only affecting the Value "Instant". 
Its affecting all of the EDGE second Values wich are grater than 2digits... so grater than 99 in YAML (as far as i can reproduce at least) 

Values Like 

EDGE L4 0.5:1.5 

seem to be Ok

Interestingly if you now load your models back into Companion and open any of the models affected. ALL of your EDGE lines are shown correct. 

 Example YAML Snipped after Loading into Companion and Back
``` 
  10:
    func: FUNC_EDGE
    def: SH2,0,10
    delay: 0
    duration: 0
    andsw: NONE
  12:
    func: FUNC_EDGE
    def: FM2,0,560
    delay: 0
    duration: 0
    andsw: NONE
  13:
    func: FUNC_AND
    def: L13,!L9
    delay: 0
    duration: 0
    andsw: NONE

```
LS 10 is correct
LS 11 is Broken 

Backup from before Christmas:
[model1_Correct_LogicSwitches.yml.txt](https://github.com/EdgeTX/edgetx/files/7820771/model1_Correct_LogicSwitches.yml.txt)

Actual File from TX after Loading into Companion:
[model1_Broken_LogicSwitches.yml.txt](https://github.com/EdgeTX/edgetx/files/7820772/model1_Broken_LogicSwitches.yml.txt)

pfeerick, 2022-01-06:
Probably related - if looking at the the hawkeyed_fixed etx model 2 (have a look at L03/L04 and AND values):

[image]

Is the same with 2.6.0-RC2 companion/simu on linux, and on Windows with #1358 sim.

Nerenye, 2022-01-06:
so the switch is incremented aswell?

PR #1361 (2022-01-06): Fix EDGE logical switch (de-)serialization
Resolves #1359
commit c7e7443c (2022-01-06): Fix EDGE logical switch (de-)serialization (#1361)

Resolves #1359
commit 0a540e8b (2022-01-07): Fix EDGE logical switch (de-)serialization (#1361)

Resolves #1359
closed 2022-01-06 by pfeerick