EdgeTX/edgetx #1365: Taranis X9D Battery Bay Serial Port 5v Not Software Controlled

techfixpros, 2022-01-06:
B01 X9D Taranis w/ETx 2.5.0

On OTx, selecting OFF in Hardware -> Serial Port turns off the 5v output. ETx 2.5.0 provides 5v regardless what this is set to.
Without being able to turn off the 5v, the external BT module has to be unplugged to turn it off, and the cables have to be routed outside of the transmitter case.

raphaelcoeffic, 2022-01-07:
This will be tackled as part of #1089.

pfeerick, 2022-04-05:
It would appear from this video (running OTX 2.3.7), that the power to VMAIN (the power connector on the serial port) is not actually software switched on the X9D... perhaps this is a mistake on the wiki since it is exactly the same wording as used to describe the module bay BATT pin, which is software controlled?) 

Earlier in the video he describes how he makes up the connector, and wires a regulator up to drop the voltage for the bluetooth module, but at this point he has the bluetooth module running, yet the serial port is still set to off, and is about to go into the settings and change it to telem mirror. 
https://youtu.be/DY8W6bd7UQk?t=846

closed 2023-05-17 by gagarinlg