EdgeTX/edgetx #1366: Power on delay doesn't work in 2.6rc2

mituritsyn, 2022-01-07:
couldn't find similar issues

**Describe the bug**
Power on delay doesn't work in 2.6rc2 any value on Radio setup tab will lead to same delay on power up
Probably caused by eeprom changes and may be fixed here: https://github.com/EdgeTX/edgetx/blob/c7e7443c30ecd89ac469ae4ed096fd30b67b0ab5/radio/src/opentx.cpp#L1655

**To Reproduce**
change power on delay on Radio setup tab to 0
power off
power on

**Expected behavior**
need to hold power on button less than a second

**Screenshots**
not applicable

**General information**
 - EdgeTX version 2.6rc2
 - Taranis qx7 transmitter

pfeerick, 2022-01-07:
This comment also mentions that, but it was in the wrong place, so thanks for the issue ;) 

https://github.com/EdgeTX/edgetx/issues/237#issuecomment-1007010969

raphaelcoeffic, 2022-01-07:
This one is a bit more complex, as it reaches out to “when do we read the radio config” and “how fast is it really”. We might have to write that specific setting in eeprom so that it can be read fast.

raphaelcoeffic, 2022-01-07:
The issue is basically here:
https://github.com/EdgeTX/edgetx/blob/eb90b4a68dc8fdd7addd6347aaaee8b3f5d50307/radio/src/opentx.cpp#L1751-L1766

`storageReadRadioSettings()` is not called anymore at this point, so that the settings are all `0`, thus leading to using the default value. However, we cannot "just" read the settings at this point, as we need to make sure we don't trigger any conversion or anything like that at this point. So we need a function to read the settings as swiftly and silently as possible (thus ignoring any possible errors) so that it is available to drive the start animation properly.

closed 2022-01-10 by pfeerick

Sekilsgs2, 2022-01-25:
This not helping - on X-Lite Pro bug present with latest release!

pfeerick, 2022-01-25:
Are you using the right firmware, as another user has reported it is working properly. X-Lite uses the X-Lite S firmware, not the X-Lite firmware.

Sekilsgs2, 2022-01-25:
> Are you using the right firmware, as another user has reported it is working properly. X-Lite Pro uses the X-Lite S firmware, not the X-Lite firmware.

Thank - i'm trying and reply you - this is not my radio (i'm on TBS MAMBO + opentx :) Edgte tx not working - current crossfire FW unsuported freertos.. ).

Sekilsgs2, 2022-01-25:
Man say that delay working while power off (no delays), but not working while power on - delay 2s present.
From radio.yml i'm see that config have right values (2-2 == 0):

`
view: 20
fai: 0
beepMode: mode_nokeys
alarmsFlash: 1
disableMemoryWarning: 0
disableAlarmWarning: 0
stickMode: 1
timezone: 0
adjustRTC: 0
inactivityTimer: 10
telemetryBaudrate: 1
splashMode: 3
hapticMode: mode_nokeys
switchesDelay: 0
lightAutoOff: 3
templateSetup: 17
PPM_Multiplier: 0
hapticLength: 2
beepLength: 1
hapticStrength: 2
gpsFormat: 0
speakerPitch: 195
speakerVolume: 12
vBatMin: 66
vBatMax: 84
backlightBright: 50
globalTimer: 453804
bluetoothBaudrate: 0
bluetoothMode: TRAINER
countryCode: 0
pwrOnSpeed: 2
pwrOffSpeed: 2
imperial: 0
jitterFilter: 1
disableRssiPoweroffAlarm: 0
USBMode: 0
jackMode: 0
ttsLanguage: "en"
beepVolume: 2
wavVolume: 3
varioVolume: 3
backgroundVolume: 3
varioPitch: 0
varioRange: 0
varioRepeat: 0
auxSerialMode: MODE_NONE
potsConfig: 
`
FW version:
https://drive.google.com/file/d/1spar0Wws4hIenxaEoj3d0UR7hj7yX6a9/view?usp=sharing

Video:
https://drive.google.com/file/d/1lG0EaAO_rYVmF4axa8ow16i3tCLBJ84C/view?usp=sharing

Sekilsgs2, 2022-01-25:
This is because his eeprom_ver == 220 but need 221 for YAML - this is check in 
```
inline uint32_t PWR_PRESS_DURATION_MIN()
{
  if (g_eeGeneral.version != EEPROM_VER)
    return 200;

  return (2 - g_eeGeneral.pwrOnSpeed) * 100;
}
```

mituritsyn, 2022-01-25:
Just updated from 2.5 to 2.6 and I have to mannually increment verion in radio.yml from 220 to 221 to get power on delay to work

Sekilsgs2, 2022-01-25:
> Just updated from 2.5 to 2.6 and I have to mannually increment verion in radio.yml from 220 to 221 to get power on delay to work

Yes i'm did this too.  but this not normal way for all peoples - need using some converter in software.

PR #1371 (2022-01-10): Load radio settings before power on delay is evaluated
This PR reads the radio settings early enough so that the power on delay can be properly evaluated. However, this introduces an additional delay (mount SD card + read settings) which is quite noticeable.

A better implementation should use settings in EEPROM / FLASH to get rid of this delay. Similar techniques could be used for some settings which are needed early on color radios.

Resolves #1366
commit 9afae09f (2022-01-10): Load radio settings before power on delay is evaluated

Resolves #1366
commit dec05bd6 (2022-01-10): Load radio settings before power on delay is evaluated

Resolves #1366