EdgeTX/edgetx #1369: No access to the internal modul in 2.6 RC2

weixelgeist, 2022-01-07:
On my Radiomaster TX16S I cannot switch on the Internal RF in 2.6 RC2
In the mode scroll down menu is only one option: OFF
In the External RF scroll down menu all options are present

[image]

pfeerick, 2022-01-07:
Try the following and see if that fixes it for you:

https://user-images.githubusercontent.com/5500713/148522714-1129777c-fb3c-441f-b354-1950ee3018cf.mp4

raphaelcoeffic, 2022-01-07:
@weixelgeist where did you update from? Some nightly build, 2.5, or something else?

weixelgeist, 2022-01-07:
The following did fix it:
Radio Settings => Hardware tab => Internal module => MULTI
Thank you

closed 2022-01-07 by weixelgeist