EdgeTX/edgetx #1373: Cannot read Models and Settings from Radio with Companion 2.6

weixelgeist, 2022-01-07:
Radiomaster 16TX with Edge 2.6.0-RC2
Win11 with Companion 2.6 
When I want to Read Models and Settings from the Radio
I get an error message: Failed to read Models and Settings from F:/
[image]

koaxheli, 2022-01-07:
Hello, I had the same problem and was able to find the error.
Two of my models had errors in the global functions.
Probably caused by the compilation from OTX to EDGE.
With Companion 2.5 everything OK with v2.6.
Changed in the transmitter, then the transmitter could be read in.

pfeerick, 2022-01-08:
Probably in the next day or so Companion should give a more helpful error message, and actually tell you what it is stuck on, so you can fix it ;)

weixelgeist, 2022-01-08:
> Hello, I had the same problem and was able to find the error. Two of my models had errors in the global functions. Probably caused by the compilation from OTX to EDGE. With Companion 2.5 everything OK with v2.6. error Changed in the transmitter, then the transmitter could be read in.

But the Global Functions are not saved in the models, they are saved once in the radio settings

pfeerick, 2022-01-08:
Probably means special functions ;)

weixelgeist, 2022-01-08:
The problem persists if you have a clean transmitter with no saved model on it

Tiefflieger68, 2022-01-08:
same issue with my TX16s.
Tested with Win10 an Win11. Both PC and used cables works well with Companion 2.3 and 2.5

weixelgeist, 2022-01-08:
It is the same on my Mac with macOS Catalina 10.15.7

Jim-n6sxb, 2022-01-09:
I'm having the same issue with Win 10, RadioMaster TX16s with EdgeTx 2.6.0-rc2.  I've tried all kinds of ways to get it to perform Read Models and Settings to no avail.  Companion will Read Models and Settings only after a Write Models and Settings.

pfeerick, 2022-01-09:
As in interim measure, until Companion actually tells you what file is causing it problems, if you go into `Settings -> Settings ...` (or the gear cog icon on the toolbar), go into the `Application Settings` tab, and down the bottom, turn on Debug Output Logging for "Application (Companion/Simulator)", and set the output logs folder to somewhere you can find it. You may need to restart Companion for the logs to start writing properly. Then, try reading from your radio again. When it shows the error, go to that logs folder, and open the `CompanionDebug_datestampe.log` file that is present, and look at the last few lines... 

[image]

i.e. in this case I have a problem with `model1.yml`. In most cases, you can simply open that model on the transmitter, make a change so that the model config file is re-written, and then undo that change. Hopefully the error is gone, and you can open them all, or it may say that it has problems with another file... in which case - open, save, try again.

weixelgeist, 2022-01-09:
This is the error message from the log file:

[D] MainWindow::readSettings():915 - Reading models and settings into temp file:  "C:/Users/weixe/AppData/Local/Temp/9136-temp.etx"
[D] findMassstoragePath():309 - Searching for "F://RADIO"
[D] readSettingsSDCard():198 - Searching for SD card, found "F:/"
[D] CategorizedStorageFormat::loadYaml():231 - "Found F:/"
[D] SdcardFormat::loadFile():42 - File "F://RADIO/radio.yml" read, size: 3790
[D] YAML::convert::decode():248 - Settings version: 221 File flavour: tx16s Firmware flavour: "tx16s"
[D] StorageFormat::setError():79 - [sdcard] error: Can't load RADIO/radio.yml:
invalid node; first invalid key: "0"
[D] main():271 - COMPANION EXIT 0

I tried to delete the radio.yml and startet the radio
the radio.yml was built new from the transmitter, but the same error message

raphaelcoeffic, 2022-01-09:
@weixelgeist which firmware version do you have on your radio? Can you please upload the files (`/RADIO` & `/MODELS`)?
Please note that #1370 does include several changes to display better error messages that should give you a better idea of what is going wrong.

weixelgeist, 2022-01-09:
Now Radio Firmware is 2.7.0 5aa866ac
Before i tried it with 2.6.0 RC2 5b21bdb
[files.zip](https://github.com/EdgeTX/edgetx/files/7834766/files.zip)

raphaelcoeffic, 2022-01-09:
@weixelgeist What I see in your files is that `models.yml` does not contain any models:
```
- "Segelflieger":
- "Motorflieger":
- "Wings":
```

Which is not an issue in itself. The `radio.yml` loads just fine here. Only issue is that I cannot see any models, as `models.yml` is empty. If I delete `models.yml` before, I get all the models (#1370 falls back to scanning all the files):
[image]

Did you try the Companion version as compiled in #1370 ?

weixelgeist, 2022-01-09:
"Did you try the Companion version as compiled in #1370 ?"

I did not find it.
Where can I load it down?

raphaelcoeffic, 2022-01-09:
> "Did you try the Companion version as compiled in #1370 ?"
> 
> I did not find it. Where can I load it down?

On the PR page, got to “Checks” tab, select the proper workflow (left hand side), you should see “Artifacts” on the bottom of the right side.

Jim-n6sxb, 2022-01-09:
Companion 2.7.0 seems to have fixed the problem.  Only thing is that the selected model in the radio reads in as model# 1, no matter which model is selected.  Minor, but worth mentioning.

raphaelcoeffic, 2022-01-09:
> Companion 2.7.0 seems to have fixed the problem. Only thing is that the selected model in the radio reads in as model# 1, no matter which model is selected. Minor, but worth mentioning.

Which version exactly? Do you mean from the PR mentioned above, or from `main`?

> Only thing is that the selected model in the radio reads in as model# 1, no matter which model is selected. Minor, but worth mentioning.

Could you also explain what you mean with this? Maybe with a screenshot?

Jim-n6sxb, 2022-01-09:
I did load from the PR #1370 as you suggested. 
On my radio, I selected the Uproar v2 as the model I'm flying.  Then plug in the USB cable and Read Models and Settings and this is what I get.
[image]

Jim-n6sxb, 2022-01-09:
Another thing I noticed is that for my models that use the internal mm with the DSM protocol, the Failsafe Mode type and Failsafe Positions show up ok in Companion, but not in the radio's model setup screen.  All my FrySky stuff looks normal.

Jim-n6sxb, 2022-01-09:
And here is a screenshot of the Companion version I'm using.
[image]

Jim-n6sxb, 2022-01-09:
Should I open a "New Issue" for these minor goobers so you can close this issue regarding reading models and settings?

raphaelcoeffic, 2022-01-09:
> Should I open a "New Issue" for these minor goobers so you can close this issue regarding reading models and settings?

For the failsafe settings it would make sense, as it is probably related to something different.

raphaelcoeffic, 2022-01-09:
@Jim-n6sxb current model should be now fixed in #1370. By the way, I believe it can be tested by saving and reloading the `.etx` file just the same.

Jim-n6sxb, 2022-01-09:
Thanks Raphael, I opened the following issue for the failsafe issues.

Failsafe Mode and Failsafe Positions Problems #1393

weixelgeist, 2022-01-09:
Companion 2.7 has fixed the problem

raphaelcoeffic, 2022-01-09:
@pfeerick looks like this PR solves a lot of issues ;-)

closed 2022-01-10 by pfeerick