EdgeTX/edgetx #1376: Enabling custom lua scripts

k4cus, 2022-01-07:
I am using nightly build of edgetx 2.7.0. I don't see custom lua scripts page. In docs it is written:

"Please note: the firmware must be compiled with the option LUA=YES for Custom scripts to be available."

but I am downloading firmware from github, I am not compiling it by myselfe. Howto enable custom lua scripts? Is it true that I need to compile firmware on my own? Every rc user must be a programmer and compile firmware by himself? Coudn't you enable lua scripts by default, as this is default feature of opentx and edge is missing it. I need it to calculate per cell voltage. Advert claims that edgetx is what opentx is and more and better. At this point I see that edgetx is missing one of important functions that opentx has build in.
Tell me that I am wrong :)

piotrva, 2022-01-07:
Custom lua scipts are enabled in pre-built releases.

pfeerick, 2022-01-07:
That documentation is now out of date - it is no longer abusing LUA as a ternary option so LUA=YES is now LUA_MIXER=Y. Can you give a link to documentation so it can be updated. 

"Custom lua scripts" is not Lua, just  as the Lua checkbox in Companion did not actually turn Lua on or off. That only turned on or off the custom lua mixer scripts, which was not on by default. Did you have a lua script that you used to add an entry for on the 'custom lua scripts' page? If so, that means you need a custom build, as we don't have the build server setup yet that opentx does (which you are using when you ticked the different options in Companion). 

It doesn't necessary mean you need to be a programmer... in the interim period until the online build server is up, you can always REQUEST a build and someone will likely build it for you, or you can use [gitpod](https://gitpod.io/#https://github.com/edgetx/edgetx/tree/main) to do it yourself if you don't want to follow any of the other guides in the wiki. 

> Custom lua scipts are enabled in pre-built releases. 

? Not sure about that... LUA_MIXER is still off AFAIK...

piotrva, 2022-01-07:
> ? Not sure about that... LUA_MIXER is still off AFAIK...

Yup, you might be right, as only luac is enabled in build I did recently.

k4cus, 2022-01-08:
Link to docs is here: 
https://luadoc.edgetx.org/part_i_-_script_type_overview/mix

Ok, so I understand that to enable lua mixers I need to compile on my own. I will go without lua mixer scirpts for now, until it is enabled by developers. Otherwise I would need to compile every time I want to update. It is not worth the effort, better I will use this time to fly my models :)
Thanks for clarification . Edgetx is really cool thing. I love that I can switch my models easily with touch screen. However I would do two things in the interface:
First, when I open model list and then I want to select model I need to click and then precisely click "select model". It is very easy to klik create or copy model, which is irritating. I am creating new models very rare and this option can be hidden somewhere.
Second thing, I would implement a lock, like in apple computers, so by default you cannot edit your model accidentally.  This is a major issue to me, that when using radio I can enter into options easily by accident. In apple stuff I have seen that user must click padlock picture, then confirm entering edit or admin mode and then modifying options is allowed. I think this would be great feature.

Combinacijus, 2022-05-23:
@k4cus 
General tutorial how to build custom firmware could be found here: <https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod>

But I'll write simple tutorial because it took a few hours to figure stuff out meanwhile whole process takes 10-15min:

1. Go to Gitpod <https://gitpod.io/#https://github.com/edgetx/edgetx/tree/v2.7.1>  (v2.7.1 tag could be changed to current version)
2. In terminal (bottom right pane) run commands (for TX16S radio for other radios see [here](https://github.com/EdgeTX/edgetx/blob/b843b715e3c6b71157ca357147f9ea567861d28d/tools/build-gh.sh#L91-L165)):
  2.1.  `cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_RULE_MESSAGES=OFF -Wno-dev -DPCB=X10 -DHELI=NO -DLUA_MIXER=YES ../`  to paste press `ctrl+shift+v`
    [image]
  2.2. [OPTIONAL] Other flags could be added [wiki (might be not up to date)](https://github.com/EdgeTX/edgetx/wiki/Compilation-options)
  2.3.  ```make -j`nproc` firmware```   or go to tutorial if using 2.8 or later versions because there's more commands
3. `firmware.bin` file in `build` folder will be generated. Download it to your computer
[image]
4. Open EdgeTX Buddy <https://buddy.edgetx.org/#/flash>
5. At the top left corner select `Local file` and select firmware.bin
6. Continue with pairing and flashing. This video might be useful: <https://youtu.be/zvakNqu_5Ec?t=293>
7. Done

I use custom built 2.7.1 firmware and Lua Mixer tab is shown now it's not that hard to build custom firmware when you done it once

ahmad-kassem, 2022-08-26:
Hello, i tried to make my custom build as per the above instructions but i am getting the below two errors when running the (make -j`nproc` firmware) command.

gitpod@edgetx-edgetx-h96nwb8hh50:/workspace/edgetx/build$ make -j`nproc` firmware
[  0%] Built target x10_alpha_bitmaps
[  0%] Built target truetype_font_24
[  0%] Built target truetype_font_9
[  0%] Built target x10_alpha_calibration_bitmaps
[  0%] Built target truetype_font_16
[  3%] Built target truetype_font_13
[  3%] Built target truetype_font_bold17
[  3%] Built target truetype_font_16en
[  3%] Built target truetype_font_17en
[  3%] Built target truetype_font_15
[  3%] Built target truetype_font_bold16
[  3%] Built target truetype_font_32
[  3%] Built target x10_fonts
[  3%] Built target x10_calibration_bitmaps
[  3%] Built target truetype_font_17
[  3%] Built target x10_splash
[  3%] Built target truetype_font_64
[  3%] Built target x10_splash_chr
[  6%] Built target x10_bootloader_icons
[  9%] Built target stm32cube_ll
[  9%] Built target x10_button_bitmaps
[  9%] Built target x10_bootloader_bitmaps
[  9%] Built target x10_themes_alpha
[  9%] Built target truetype_fonts
[  9%] Built target datacopy
Version 2.7.1
Input file lua_exports_t16.txt
Output file lua_exports_t16.inc
Documentation file lua_fields_t16.txt
Generated 40 items in luaFields[]
Generated 7 items in luaMultipleFields[]
[  9%] Built target lua_export_t16
[  9%] Built target x10_slider_masks
[  9%] Built target x10_volume_masks
[  9%] Built target x10_layouts_masks
[ 18%] Built target x10_masks
[ 34%] Built target x10_themes_masks
[ 34%] Built target x10_bitmaps
[ 34%] Built target radio_translations
Scanning dependencies of target bootloader
[ 43%] Built target bootloader
Scanning dependencies of target firmware
_**_In file included from /workspace/edgetx/radio/src/gui/colorlcd/preview_window.cpp:21:
/workspace/edgetx/radio/src/gui/colorlcd/preview_window.h: In member function 'void PreviewWindow::drawTime(BitmapBuffer*)':
/workspace/edgetx/radio/src/gui/colorlcd/preview_window.h:290:20: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
  290 |     if (s > 0 && s < sizeof(str) - 6 /* 00:00\0 */) {
      |                  ~~^~~~~~~~~~~~~~~~~
In file included from /workspace/edgetx/radio/src/gui/colorlcd/radio_theme.cpp:36:
/workspace/edgetx/radio/src/gui/colorlcd/preview_window.h: In member function 'void PreviewWindow::drawTime(BitmapBuffer*)':
/workspace/edgetx/radio/src/gui/colorlcd/preview_window.h:290:20: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
  290 |     if (s > 0 && s < sizeof(str) - 6 /* 00:00\0 */) {
      |                  ~~^~~~~~~~~~~~~~~~~_**_
[100%] Built target firmware
gitpod@edgetx-edgetx-h96nwb8hh50:/workspace/edgetx/build$ 

Any idea hot to resolve this problem?

piotrva, 2022-08-26:
It seems like you get only warnings and the build finished successfully.

ahmad-kassem, 2022-08-26:
the build works, thanks, but i don't know if these error are critical to the radio working or not, what do you think?

piotrva, 2022-08-26:
No, they are not critical in any means.
And for such questions I strongly encourage you to join Discord server (or FB group) - GitHub should be only for very technical and problem-oriented stuff.
Links to join mentioned groups can be found here: http://edgetx.org/