EdgeTX/edgetx #1378: Model Notes

koaxheli, 2022-01-07:
Radiomaster TX16S

The checklist display is in Companion v2.6.
It was still available in v2.5.
[image]
[image]

**General information**
EdgeTX "Santa" 2.6.0-RC2

pfeerick, 2022-01-07:
Um... so what is the problem? :grin:

koaxheli, 2022-01-08:
Radiomaster TX16S

That the checklist display in Companion v2.6. is missing, is not displayed.

pfeerick, 2022-01-08:
Thanks... the difference a _missing_ word makes! :)

Right, so with a model named `fpv`, and model notes named `fpv.txt`, it seems like simulator is looking for htis file... :-O

`f_open(E:/Documents/EdgeTX/RadioMaster TX16s/SD Card V2/SOUNDS/en/fpv/name.wav)`

although companion knows perfectly well it is in 

[image]

I also don't recommend using the sd browser to find that file instead, unless you *want* simulator and companion to crash.

elecpower, 2022-01-08:
> Right, so with a model named `fpv`, and model notes named `fpv.txt`, it seems like simulator is looking for this file... :-O
> 
> `f_open(E:/Documents/EdgeTX/RadioMaster TX16s/SD Card V2/SOUNDS/en/fpv/name.wav)`

Simulator and radio always look for this file /model name/name.wav as a audio call out on radio boot. I use this feature.

pfeerick, 2022-01-08:
Ok, so trying again, and it now goes beyond that - maybe I scrolled back and missed the lines after...

First at around 8 seconds is the attempt to view the model notes via the menu, and then at around 18s trying to view it via sd browser and eventual crash. 

[FirmwareDebug_22-01-08_21-18-04.zip](https://github.com/EdgeTX/edgetx/files/7833117/FirmwareDebug_22-01-08_21-18-04.zip)

elecpower, 2022-01-08:
at 18s f_open(E:/Documents/EdgeTX/RadioMaster TX16s/SD Card V2/C:/Users/pfeer/AppData/Local/Temp/etx-CfELgo/MODELS/fpv.txt) = INVALID_NAME (FIL 000000d62f7fedf0)

So combines both directories and appends file name so something definitely wrong with sd browser in simulator mode. Also it should halt after 1st failed attempt.

elecpower, 2022-01-08:
From the log, the radio start up doesn't bother to look at the profile sd card for the checklist file. Checklist is displayed on radio start up for B&W radios tested with X9D+ simulator.
There is different logic between deciding whether the Model Notes icon should be shown and actually trying to find and read the checklist file on clicking the icon.

pfeerick, 2022-01-09:
There are three ways to look at this at - you can enable the checklist (which I didn't), and you can view the model notes directly (which I did). If you enable the checklist it doesn't show up. If you attempt to view the model notes it doesn't show up. If you attempt to view the model notes through the sd browser, and it explodes. 

At the end of the day, something is wrong with the path, which is what is setting everything off.

elecpower, 2022-01-09:
When the target radios are compiled in simu mode any file access code needs to check both sd paths (primary and if not found then profile).
Depending on how the libsim is started Companion (etx) or Simulator (data file, data folder, sd card) depends on what is in the primary path.
It is not practical to have one merged path, as every file in the profile path would have to be copied to the temp path on every simulator launch (hot beverage time).

piotrva, 2022-01-12:
Yup, reading a file using SD browser crashes the sim, and notes are not being displayed.
It gets into infinite loop of opening the file:
`93.14s: f_open(/src/build-tx16/SDCARD_SIM/tmp/etx-lDHiie/MODELS/New_model.txt) = INVALID_NAME (FIL 0x7fe21a8c5690)
93.14s:         not found`
That does not stop even when simulator is closed (or it is just so much terminal data that it keeps going after simulation is disabled...
Do you know if it is a thing on other colorlcd radios? As this stops me for now on preparing #1417 for them...

pfeerick, 2022-01-13:
Right, so this is something very specific to the data file / launch through Companion mode - as if you run the simulator standalone, and instead use the data folder or sd image modes, everything works properly.

pfeerick, 2022-01-13:
What I found rather odd was this... when in data file mode, it correctly identified the existence of the file, but then changed into the tempoary unzip directory for ETX data file. btw, if you shove the model notes into that folder, that works also... 
```
207.54s: f_stat(C:/ETX/SD/MODELS/ParkMaster.txt) = OK
207.54s: [ViewMain] Focus lost
208.40s: [ViewMain] Focus gained
208.40s: f_chdir(C:/Users/pfeer/AppData/Local/Temp/etx-zkwtwq/MODELS)
208.40s: 	not found
208.40s: f_stat(ParkMaster.txt) = error 2 (No such file or directory)
208.40s: 	not found
208.40s: f_stat(ParkMaster.txt) = error 2 (No such file or directory)
208.40s: 	not found
208.40s: f_stat(model1.txt) = error 2 (No such file or directory)
208.40s: 	not found
208.40s: f_stat(model1.txt) = error 2 (No such file or directory)
```

piotrva, 2022-01-16:
@pfeerick it is not odd, but intentional, as editing things in Companion does not mess up with backed-up contents of SD-Card on your drive - instead it creates a "tmp" and sets radio.yml and modelXX.yml there based on actual settings in Companion. Issue is that it should symlink or copy there matching checklist files.

piotrva, 2022-01-16:
And I do confirm that making a copy of requested files:
`root@632a9d1ebdc8:/tmp/etx-fcyfGF/MODELS# cp /src/build-tx16/SDCARD_SIM/MODELS/New_model.txt New_model.txt`
allows to bypass a problem for now ;)

piotrva, 2022-01-16:
As for fix for development purposes one may use following script, after correcting paths: 
[fix_notes.sh](https://github.com/EdgeTX/edgetx/files/7878074/fix_notes.sh.txt)

pfeerick, 2022-01-18:
> @pfeerick it is not odd, but intentional, as editing things in Companion does not mess up with backed-up contents of SD-Card on your drive - instead it creates a "tmp" and sets radio.yml and [...]

No, I don't think that is quite right - have another read of [what Neil said above](https://github.com/EdgeTX/edgetx/issues/1378#issuecomment-1008208618) - there are *two* paths it is supposed to use, and to me that log is suggesting that for some reason it is switching paths even though the first path was correct. Either that, or maybe it is switching paths to read something, and not setting it back.

pfeerick, 2022-08-19:
@eshifri Can you play with this one... I've finally spotted the actual culprit (and who was responsible... but I won't name names :P) ...

Basically getting rid of this, and using a full, rather than a relative path, somewhat similar to what the B&W radios do, will resolve this... but I've not been able to get it completely working again... 😆 

 https://github.com/EdgeTX/edgetx/blob/6648dffc8c89d34fd26aba544e507d3cf3b20e47/radio/src/gui/colorlcd/view_text.cpp#L214

eshifri, 2022-08-19:
Sorry, I'm a little bit at a loss...
The problem is not related to the lvgl fix (referenced above), correct? This code is much older.
Is the problem in the Simulator or in Companion? Looks like Simulator is working for me.
If it is in Companion, does the Companion use this file?

Or, in other words: how do I reproduce the problem? :-)

pfeerick, 2022-08-19:
Yes,  this is indeed the issue raised. Basically,  it is in two parts - if you have the show checklist option enabled,  the model notes will *not* be displayed on colorlcd,  it seems due to the chdir call confusing simu. And while the view notes icon will show,  in the menu,  it will fail to actually do anything, for the same reason.

I. E.  With a colorlcd profile,  create a sd card path with a model notes file,  and then configure a model to show preflight checklist and try it in companion simulator. It won't show.  And then try to view model notes via the main menu.  On radio this is all fine,  just simulator that is confused due to the dual path handling it does. :(  B&W model notes path handling works fine in simulator also.  ;)

PR #2223 (2022-08-25): Fix full path to model notes for CP Simulator (fix #1378)
Fix access to the Model Notes from Companion Simulator

Fixes #1378

Summary of changes: use full path instead of chdir() to test for file existence.
Only two file operations are taking care of the differences between TX and computer: f_stat() and f_open().
closed 2022-08-25 by pfeerick