EdgeTX/edgetx #1382: X9D+ models incorrect Internal RF/protocol choices

elecpower, 2022-01-07:
**Describe the bug**
When editing a model setup the only available choices are OFF and Multi in radio sim and OFF in Companion.
Note: this radio does not have Internal module support in hardware settings.

**To Reproduce**
Edit a model and try to change the Internal RF (radio) or Internal Module Protocol (Companion).

**Expected behavior**
Choices appropriate for the radio.

**Screenshots**
[image]

**General information**
 - EdgeTX version 2.6 main
 - Hardware simulator Frsky X9D+
 - Companion version 2.6 main

pfeerick, 2022-01-08:
I've having trouble reproducing, but I'm using the #1342 Companion/Simulator if it is version specific - I've tried X9D+2019, X9D+ and X9D and all seem to be correctly going through OFF, ACCESS/ACCST,XJT as options?

elecpower, 2022-01-08:
By bad, the radio sim behaves after I performed a clean build of everything. Can only assume a cached entry.
However Companion is not functioning as expected so up to me to fix.

closed 2022-01-10 by pfeerick