EdgeTX/edgetx #1384: 2.5 => 2.6 Yaml Companion

pfeerick, 2022-01-08:
When reading from radio (TX16s) with a build from #1360 on Windows I noticed the following when working from the bin files instead of yaml (I had deleted all the ymls in MODELS and RADIO)
- Reading was fine
- Writing (withing making any changes) wrote all the models files to yaml

But:
- `RADIO\models.txt` was updated to point to the yaml files, instead of writing to `MODELS\models.yml`
- `RADIO\radio.yml` was not written.
- Making changes to radio settings worked, but they continued to be written to `RADIO\radio.bin`

raphaelcoeffic, 2022-01-08:
@pfeerick were you writing to a `.otx` file, or `.etx`?

pfeerick, 2022-01-08:
Whatever Companion generated by default when reading from the radio with nothing open, so will have to check that.

raphaelcoeffic, 2022-01-09:
@pfeerick I just tried with #1370 and it works as expected. I cannot tell you which part really fixes it though.

pfeerick, 2022-01-09:
Ok, so reading is unlocked now... but I think I've more correctly identified the symptoms (but not the cause yet...). When the latest version of #1370, it once again reads .BIN files (i.e. remember, this is a 2.5 => 2.6 conversion, I've deleted the yml files) just fine, into a new `.etx`. 

But, when I then write that... 
- it writes yml model files which have bin content
- changes `model.txt` to point to yml files
- doesn't write `models.yml` in MODELS folder

the 2.6.0 companion build I'm running does the same so it's not something introduced in #1370.

PR #1402 (2022-01-10): [companion] fix: only write yaml to radio, not bin
Fixes #1384

- Fixes issue whereby Companion was writing bin files with yml extension
closed 2022-01-10 by pfeerick