EdgeTX/edgetx #1385: Output Screens Enhancement

rburrow87, 2022-01-08:
This is for the color screen radios.

It would be nice if the Outputs screen had something to indicate channel positions and/or movement. OpenTX has a little arrow that points towards the min or max depending on position, or a dash if at center. Makes it a bit easier than flipping back and forth from the mixes page.

[image]

Having the current us value on each row would be nice too, or do it like the B&W radios and show it in the bar that says "OUTPUTS" for whichever row is highlighted.

The B&W radios also have a feature where if you long press on an output, it offers an option for "copy the min/max/center to all." Would be nice to see this on the color radios too.

micromega1966, 2022-01-08:
Era già stato chiesto qualche tempo fa con la versione 2.5 ma non è stato preso in considerazione. Peccato, sarebbe molto utile.

natiho2, 2022-01-13:
I'm also surprised they do not address this,
it exists in every transmitter,
It feels like a body that is missing a very useful item..
In the meantime I went back to opentx until it fixed
If it will be fixed at all.

raphaelcoeffic, 2022-01-13:
> I'm also surprised they do not address this, it exists in every transmitter, It feels like a body that is missing a very useful item.. 

We have all only 2 hands... But no worry, we know this is an issue.

> If it will be fixed at all.

Don't be rude 😅 You are very welcome to participate and help make things faster.

MervDale, 2022-01-13:
> I'm also surprised they do not address this, it exists in every transmitter, It feels like a body that is missing a very useful item.. In the meantime I went back to opentx until it fixed If it will be fixed at all.

As in OpenTx you can name the Output channel(s) i.e. Channel 1 = Rudder,  if you are not sure which is which. 
 INV tells you if  you have reversed (Inverted) the channel so  OUTPUTS gives you pretty much everything OpenTx does except for two little arrows and a dash that tells you which way you are moving the stick,   I find it a bit odd if you're not sure of that !

natiho2, 2022-01-13:
> > > I'm also surprised they do not address this, it exists in every transmitter, It feels like a body that is missing a very useful item.. In the meantime I went back to opentx until it fixed If it will be fixed at all.
> > 
> > 
> > As in OpenTx you can name the Output channel(s) i.e. Channel 1 = Rudder, if you are not sure which is which. INV tells you if you have reversed (Inverted) the channel so OUTPUTS gives you pretty much everything OpenTx does except for two little arrows and a dash that tells you which way you are moving the stick, I find it a bit odd if you're not sure of that !
> 
> and the value the output has at the moment. It´s actually really helpful when setting up big gliders for example.

You can also fly with a 4-channel remote control, you can also drive a car with mechanical window handles, etc.
I really do not understand the response you wrote
The two little arrows you mentioned are what make all the difference to an efficient setup

natiho2, 2022-01-13:
> I'm also surprised they do not address this, it exists in every transmitter, It feels like a body that is missing a very useful item.. In the meantime I went back to opentx until it fixed If it will be fixed at all.
> 
> I really do not understand the response you wrote
> 
> Maybe you should stay with OpenTx then and can also look for the manners you seemed to have dropped on the way 😉

I'm sorry if you were hurt I did not mean that,
I mostly fly helicopters and there is no room for mistakes as you know
When you setup a helicopter the arrow on the output page is very helpful and true that it is possible even without it, but it mostly complicates things, after all we all switched to edgetx
Because it's a touch screen and program things faster and more efficiently.

PR #1664 (2022-03-11): Add channel bar to Output Edit Screen
Resolves #1385

Summary
- [x] Add channel bar to header of outputs edit page
- [x] Add values of each channel in to the end of row in overview table on wide lcd screens (not enough space on NV14, so no change there). Added commented out code that can show in % instead.
- [ ] ~~Add value of current focused channel of narrow screen radios - I think I just figured out how to track focused channel~~
This can wait until 2.8 along with adding the footer statusbar for mixes and outputs page. 

TX16S:
[image]
[image]

NV14:
[image]
[image]
commit 9b830ac1 (2022-03-11): feat: colorlcd - Add channel bar to Output Edit Screen and us values to outputs table on wide displays (#1664)

* feat: Add channel bar to Output Edit Screen
Resolves #1385

* Wide colorlcd, show channel val on outputs table
I've also add commented code for those who want percentage instead of us
closed 2022-03-11 by pfeerick