EdgeTX/edgetx #1389: 6-Pos Switch is not working on Radiomaster TX16s

onki69, 2022-01-08:
I am using the current 2.6 nightly (Jan 8th) build with a Radiomaster TX16s.
The 6 pos  switch is udes in some of my models to switch flight modes.
It worked well on 2.5 but there is no response at all with 2.6.

I can see the switch light is moving from 2 to 3 when I press the corresponding switch but the switch itself is not recognized by the OS.
It means when I try to setup a flight mode nad want to assign a switch there is no action when I press one of the 6-pos switch.
If I select the switch from the list and want to switch later the flight mode still no response at all.

I thought there was a hardware problem but when I downgrade to 2.5 everything works well.

eshifri, 2022-01-08:
Works for me in the latest code.
Can you please check the 6POS is calibrated?

pfeerick, 2022-01-08:
6POS is a analog input also, so along with your stick, sliders and pots, you need to center and move to the extremes of each input. In the case of the 6POS, I pick 3 or 4, and then press each button in turn from 1 to 6, and then 6 through to 1, with a slight pause between each.

onki69, 2022-01-09:
OK - didn't knoiw that the 6 pos is analog.
After calibration it works.
But since all other inputs were OK it seems that the BIN -> Yaml conversion is faulty at this point as the 6 pos calibration was not transferred as it worked with 2.5.

raphaelcoeffic, 2022-01-09:
>  it seems that the BIN -> Yaml conversion is faulty at this point as the 6 pos calibration was not transferred as it worked with 2.5

Can you please upload the binary files? At best, just upload the complete content on `/RADIO` & `/MODELS` zipped together.

raphaelcoeffic, 2022-01-10:
@onki69 I cannot reproduce either.

Here is what I did:
- install 2.5.0 and wipe storage
- start, calibrate and verify 6-pos switch is working properly
- update to latest `2.6`, thus triggering storage update
- verified that 6-pos still works perfectly

closed 2022-11-06 by pfeerick