EdgeTX/edgetx #1390: Make it possible to switch at runtime between bluetooth and serial2 on TX16s (and others)

wimalopaan, 2022-01-08:
When using bluetooth (BLUETOOTH must be compiled in) the folwoing situation holds:

- serial2 disappers (as it is the usart for the BT module
- when setting BT to Trainer/Telemety the EN pin of the BT module is enabled
- when setting BT off, the EN is disabled.

When disabling the BT, the TX of the BT goes into floating state. And on other BT modules this is even an OpenDrain output.

Therefore the BLUETOOTH and serial2 could be compiled in at the same time. The BT is enabled with EN = high. An external device should send only if EN = CTR = LOW.

This would give the most flexiblility.
For serial2 there must be an additional runtime option Bluetooth in addition to SBus, LUA, Telemetry, ...

wimalopaan, 2022-01-08:
Maybe considered in #1089

brainbubblersbest, 2022-01-09:
Nice. So internal bt gets automatically Disabled after change to aux Serial? So internal bt doesnt needs to be removed? Beautifull. I desoldered bt on my x12 to get a Free Serial. +1 for that

wimalopaan, 2022-01-09:
@rotorman corrected me, that the above is just the other way round: internal BT module has !EN input, so it is enabled when EN/CTR is low. Therefore the external device should be enabled with EN/CTR high.

pfeerick, 2022-01-09:
You can edit your own posts, so it might be a good idea to update the first post with the correct info ;)

wimalopaan, 2022-01-09:
> You can edit your own posts, so it might be a good idea to update the first post with the correct info ;)

Edited