EdgeTX/edgetx #1393: Failsafe Mode and Failsafe Positions Problems

Jim-n6sxb, 2022-01-09:
**Describe the bug**
Two issues
1. The Failsafe Mode and Failsafe Settings do not appear in the Model Settings screen on my radio for Internal mm RF module set to the DSM protocol.  They are displayed in the Companion Model Setup screens for these models read from the radio.  The radio does display these items in the model setup screens for my FrSky protocol models.  I only have DSM and FrSky receivers.
2. The Failsafe Settings are all set back to zero whether I change them to non-zero values in the radio and read them back to Companion or change them in Companion and write them to the radio. PILOTS TAKE NOTE BEFORE FLYING WITH 2.6.0-RC2.

**General information**
  - EdgeTX 2.6.0-rc2
  - Companion Version 2.7.0 "dev", Jan 9 2022
  - Windows 10 - latest rev
  - RadioMaster TX16s

pfeerick, 2022-01-10:
This sounds to be all companion - related. as
1) DSM does not have configurable failsafe options - only at the receiver level, so companion shouldn't be showing those options. But to be fair, this is exactly the same as with OpenTX version of companion - it doesn't discriminate between protocols that have do or dont have failsafe options, support low power rf, do channel mapping, etc. So I think this is more a enhancement for Companion. 
2) That is a definite no-no... it shouldn't be changing.

Jim-n6sxb, 2022-01-10:
@pfeerick - I'm going to go through my models with a fine tooth comb to check for funny business as a result, I believe, of converting from 2.5.0 to 2.6.0 just to be sure that I don't have a bunch of folks chasing mysteries as in issue #1392 that I opened.

Jim-n6sxb, 2022-01-10:
Very much further testing shows that indeed Failsafe Settings values are being re-set to zero upon reading and writing the radio with Campanion 2.7.0.

After a manual change of the Failsafe Settings values, the values persist after power cycle of the radio.  But, when read into Companion, they show up zeroed.  In addition, when changed again in Companion and written to the radio, they reset to zero.

raphaelcoeffic, 2022-01-10:
@elecpower can you please have a look at this one?

Jim-n6sxb, 2022-01-10:
As an FYI, I did confirm this morning that this is a Companion issue only. This issue does not come up when I do all 2.5.0 to 2.6.0-rc2 model conversions within the radio.

PR #1399 (2022-01-11): Fix failsafe settings
Resolve #1393

Summary of changes:
- (de-)serialize failsafe settings properly
- fixed radio failsafe display (use proper scale instead of raw value)
commit 7a628d4c (2022-01-11): fix: Companion read/write of failsafe settings, how displayed on radio (#1399)

* Fix failsafe settings

Resolve #1393

* trigger save if failsafe numbers changed + cleanup
commit 48ef8909 (2022-01-11): fix: Companion read/write of failsafe settings, how displayed on radio (#1399)

* Fix failsafe settings

Resolve #1393

* trigger save if failsafe numbers changed + cleanup
closed 2022-01-11 by pfeerick