EdgeTX/edgetx #1394: Throttle Timers Not Working Properly

Jim-n6sxb, 2022-01-09:
As background, we have three throttle timers, Throttle, Throttle Start and Throttle %.  The way they operate in EdgeTX 2.5.0, which I find very nice (a little different than OTX) are the following;

Throttle - Doesn't start counting until some condition is True and the throttle is greater than -100, then counts by seconds (up or down) until the condition is False.

Throttle Start - Doesn't start counting until some condition is True and the throttle is greater than -100, then counts by seconds (up or down) unless the throttle stick is all the way down (-100). If the timer is paused by throttle = -100, it resumes when throttle > -100.  Stops counting when the condition is False.

Throttle % - Doesn't start counting until some condition is True and the throttle is greater than -100, then counts (up or down) by seconds proportional to throttle position.  Stops counting when the condition is False.

How the timers work on my RM TX16s radio with EdgeTX 2.6.0-rc2 are the following;

Throttle - Starts counting by seconds (up or down) as soon as the condition is True with any throttle position. Keeps counting until condition is False.

Throttle Start - Behaves just like Throttle above.

Throttle % - Behaves like the two above except it counts 1 second for every two, regardless of throttle position.

raphaelcoeffic, 2022-01-09:
This is strange as the implementation has not changed. So maybe a bug somewhere?

eshifri, 2022-01-09:
I cannot reproduce it: everything works properly in current 2.7 code.

Jim-n6sxb, 2022-01-09:
Maybe a copy of my Models and Settings would help?
[Planes 1-9-2022.zip](https://github.com/EdgeTX/edgetx/files/7835998/Planes.1-9-2022.zip)

Thermikflieger, 2022-01-09:
I can confirm what eshifri says, the throttle timers are working correctly.
Jim-n6sxb also swapped the functions "Throttle" with "Throttle Start" in his description.

Jim-n6sxb, 2022-01-09:
> I can confirm what eshifri says, the throttle timers are working correctly. Jim-n6sxb also swapped the functions "Throttle" with "Throttle Start" in his description.

My bad, I did indeed swap those two, sorry.  And I was trying to be so meticulous too.  Have you looked at my Models and settings to see if you can see anything in my setups that are causing this behavior?

Thermikflieger, 2022-01-09:
I loaded your models and tested the timer functions with "Bixler". In fact, one timer runs half as fast as the other. This is mysterious, it shouldn't be like that! You have chosen a relatively complex release of the timer via L03. I think that could be the reason for the wrongdoing. Try just a simple bat like SF instead of L03. I'll try that right away and then get back to you.

Thermikflieger, 2022-01-09:
It is not L03, but you have selected CH1: Ail as the gas timer source. Since CH1: Ail is at 0%, Timer1 "Throttle" runs at half speed (TH%), Timer 2 "Tot Time" runs normally (THt). Change the gas timer source to THR and everything is fine.

Jim-n6sxb, 2022-01-09:
Here is a file with only the Bixler 1-1 model.  I changed all the logical switch timer triggers to a simple mechanical switch.
I created a screen that shows the three timers behaving badly, just as described above (switching Throttle and Throttle Start as noted by Thermikfieger).

[Timer Test.zip](https://github.com/EdgeTX/edgetx/files/7836216/Timer.Test.zip)

Jim-n6sxb, 2022-01-09:
Funny, I saw that in earlier iterations of my playing with 2.6.0-rc2 and I swear I changed it. But somehow it switched back.  Probably during the time I was having trouble loading files from the radio before Companion 2.7.0.  I'll take a closer look, and thanks for spotting that.  
@raphaelcoeffic - Could you give me a little time to make sure that the above solves the problem, before you close this issue?  As soon as I'm convinced it was just a model conversion foopa, I'll post it here.

Jim-n6sxb, 2022-01-10:
@raphaelcoeffic @Thermikflieger - Just checked it all out and it working wonderfully.  It may still a bit of a concern in that I'm pretty sure that gas source going to chan 1 happens in the conversion from 2.5.0 to 2.6.0.

Thank you both for the work you do.  BTW, I bought a second radio, just so I can help find this stuff.  Is this the kind of help you want?  If you only want more professional coder help, my feelings won't be hurt, just let me know.

pfeerick, 2022-01-10:
> Is this the kind of help you want?

Yes, 110%. Giving the coders a hard time (testing stuff they coded to make sure it works with no ill side effects) is the most tedious and hardest part... and the more people who are testing different things, with different radios and requirements, the more bugs we can squash and eliminate :) So more the merrier! :)

closed 2022-01-10 by pfeerick

Jim-n6sxb, 2022-01-10:
Roger that, and I will do some testing to determine just that.

Jim-n6sxb, 2022-01-10:
@pfeerick I did find a model conversion problem and posted and opened a new issue #1398 for it.