EdgeTX/edgetx #1396: Model notes/checklists storage & bind with model

piotrva, 2022-01-09:
Model checklist should be better bound with model. For now it is connected by model name, and in case of model name change - you loose connection.
Also when duplicating the model and afterwards changing name - same happens.

As ETX has migrated to YAML storage - maybe this should be included in model.yml files?
Of course that would require not to load this content to RAM memory of the radio, as it is not as important as model configuration (maybe similar machaniem as that for loading only header of the model when only partial data required).

I think it is good to discuss this issue.

eshifri, 2022-01-10:
You mean explicitly include the model notes file name into the model setup?
So e.g. several models can share the same notes file?

piotrva, 2022-01-10:
I mean to include notes themselves in the model setup - to remove need of separate file.

pfeerick, 2022-01-10:
I think he means to put the model notes into yaml, but IMO having the filename in the config may be the better option... As if you break the connection, you could point it to the new name that way. And yes, as a bonus that could allow shared model notes if you have a couple models that are configured the same way.

piotrva, 2022-01-10:
It is also a good solution - but in this solution it might not be clear to which models which file is assigned to...

raphaelcoeffic, 2022-01-10:
@piotrva the main issue of adding the notes to the model file itself is that it will require some additional support in the radio YAML parser to support raw text being added. Additionally, this would considerably slow down parsing the file, as it can be potentially inflated a lot.

piotrva, 2022-01-10:
That is a point also.
So maybe rational solution would be to store file name, as @pfeerick suggested, or use model number as a reference for the file (while also preparing copy of the file during model copy) ?

pfeerick, 2022-01-10:
I think the code already looks for notes to be the same name as model file already, so perhaps part of what is needed is simply copying the notes files as well as the model config when copying a model). And/or adding storing of filename.

piotrva, 2022-01-10:
As far as I remember for main branch it still is looking for model name. Will check it.

eshifri, 2022-01-10:
For color LCD the code is looking for model name and if the file does not exist - for the file name.
I do not remember what happens for B/W, but, most likely, it is the same.
There is one problem with including notes into YAML: we will need a special editor (in Companion): there will be some users who are not comfortable with editing YAML in Notepad.

I'm not sure about copying the notes: notes are edited outside the radio, so it is more convenient to have the model name than the model file name. So copy will involve some hoops and loops when renaming the model.
I think reference to the file is more convenient.
And I think many people need the same notes for multiple models.

piotrva, 2022-01-10:
Yup, considering all above file (name) reference would be ok IMHO.
And what is most important - would be much better then reference by model name, which may be left for compatibility.

eshifri, 2022-01-11:
@piotrva Do you plan to implement this yourself? (I can handle it if you do not.)

piotrva, 2022-01-11:
@eshifri at the moment working on "clickable" checklists for B&W, maybe tomorrow will have time for that, but if you can make it - just go for it ;)