EdgeTX/edgetx #1403: Switch for Timer will be saved in .etx file but not load when reopen

ParkerEde, 2022-01-10:
I use Timer 1 and 2 with this settings (part of yml file, extracted from etx file):
timers:
  0:
    swtch: NONE
    mode: THR_REL
    name: Motor
    minuteBeep: 0
    countdownBeep: 2
    start: 210
    persistent: 1
    countdownStart: -1
    value: 208
  1:
    swtch: L64
    mode: ON
    name: Flugzeit
    minuteBeep: 0
    countdownBeep: 0
    start: 0
    persistent: 1
    countdownStart: -1
    value: 6

As you can see, for Timer 2 is the switch logical switch L64. But after reopening this etx file in companion, the Timer 2 has no switch anymore:
[image]
Same behaviour for STandalone Simulator.

Next issue is, there is no possibility to set the Timer switch with lua script. The debug output shows no switch.
1.step, I set manually Timer2 switch to L64 in simulator
2.step, I start my script to debug the output:
table: 000001a2a072cbc0 {
  [value] => 27
  [countdownBeep] => 0
  [persistent] => 1
  [name] => "Flugzeit"
  [minuteBeep] => false
  [start] => 0
  [mode] => 1
}

Switch is missing.

closed 2022-01-10 by raphaelcoeffic

ParkerEde, 2022-01-10:
Hi @raphaelcoeffic, puh, very quick - nice - thank you. 
One more question. Did you also solved the luascript part of this ticket, or should I create a new ticket?
Kind regards

raphaelcoeffic, 2022-01-10:
@ParkerEde What happens when you start the standalone simulator is the following:
- reads from whatever input it should use
- write YAML again that is then fed to firmware

What happens here is that during the read/write process, the field gets wiped, thus preventing you from seeing anything in your script. You can check by having a look at the debug output and read the YAML file given to the simulated firmware.

ParkerEde, 2022-01-10:
okay. I will check with the next upcomming companion version.
Thank you.

commit 1c8639ba (2022-01-10): Fix raw switch index checks (#1403)

Resolves #1403
commit acebd2b7 (2022-01-10): Fix raw switch index checks (#1403)

Resolves #1403