EdgeTX/edgetx #1404: model.insertMix creates 64 identical Mixerlines

ParkerEde, 2022-01-10:
This line in my luascript:
`model.insertMix(11,0, {name="STest", switch=140, speedUp=10, source=102, weight=100, curveValue=32, curveType=3})`

creates in MIXES 64 identical lines for CH12.
I'm not sure, is this a mistake by me, or is this a bug?

raphaelcoeffic, 2022-01-10:
@ParkerEde Which firmware version are using while testing? This seems caused by an issue in 2.6 RC1 which has already been fixed since then (see https://github.com/EdgeTX/edgetx/pull/1163).

ParkerEde, 2022-01-10:
I use 2.6.0 RC2 and I only use the Companion simulator and the standalone simulator.

raphaelcoeffic, 2022-01-10:
> I use 2.6.0 RC2 and I only use the Companion simulator and the standalone simulator.

ok, but without any other info, it will be non-trivial to debug. So we need the following:
- either a set of YAML files that show the display bug,
- or a full set of files including full LUA script and a list of steps to reproduce.

ParkerEde, 2022-01-10:
no problem. I will prepare and come back.... Thank you

ParkerEde, 2022-01-10:
[LUA and yml.zip](https://github.com/EdgeTX/edgetx/files/7840023/LUA.and.yml.zip)
How to reproduce:
1. open standalone simulator (DataSource: FILE)
2. create a new model (without wizard)
3. execute the attached mixes.lua (SYS/SD CARD/SCRIPTS/Mixes.lua
4. Have a look to MDL/Mixes/CH12

Also attached is the result, in my case model41.yml

ParkerEde, 2022-01-13:
Hi @raphaelcoeffic 
do you need more information or anything else to this issue?

raphaelcoeffic, 2022-01-13:
> Hi @raphaelcoeffic do you need more information or anything else to this issue?

No, sorry, just time 😌

raphaelcoeffic, 2022-01-14:
@ParkerEde Now that I actually had a look at the LUA script, it looks pretty clear: `insertMix()` is executed all the time! So basically it inserts mixes until the function starts to fail (no free mixes available).

The little detail you have been missing here is that the `run()` is executed again and again until it returns something else than 0. Here I believe you should make sure that `insertMix` is executed only once (for instance by using a variable global to your script or maybe checking the mix is already there?Not sure what the use case is).

closed 2022-01-14 by raphaelcoeffic

ParkerEde, 2022-01-14:
yes thats right. But I thought, that the line parameter is fix and only this line would be inserted?
I've modified the script, now it runs only one times, but the behaviour of the insertMix is still unintelligible.
If I start it more than one times on a new created modell, there appears more lines as should.

Are you sure, there is no bug?
Kind regards and many thanks.

[RK-StabiRx_2.3.xx.zip](https://github.com/EdgeTX/edgetx/files/7870189/RK-StabiRx_2.3.xx.zip)
.

pfeerick, 2022-01-15:
It depends on exactly what the function is actually supposed to do - it's named *insert*Mix, not setMix  - so to my mind `insertMix(11,0, ...) should be inserting into the first mixer row for channel 10, and doing so every time you run it. However, a function called `setMix(11,0, ... )` would set channel 10, mixer line 1 to ... . I suspect the intended logic may have been that you should use be getMix to validate it's the mixer line you want to edit or if changes are needed, deleteMix to clear it, and then insertMix to cleanly insert a newMix. Making it so you can't have a mix that has potentially conflicting user and lua set values unless you specifically code it that way.

ParkerEde, 2022-01-15:
Yes exactly. This is also how I did it in the script

1. Step delete all existent rows of a CH
2. Insert three new mix lines for that CH
Like this:
`clearmix(8, 0, 63)
	model.insertMix(8,0, {name="Gain9M", switch=19, source=108, weight=80, multiplex=0, speedDown=0, speedUp=0, delayDown=0, delayUp=0, carryTrim=true, offset=0, flightModes=0, curveValue=0, curveType=0})
	model.insertMix(8,1, {name="GainLd", switch=20, source=100, weight=30, multiplex=0, speedDown=0, speedUp=0, delayDown=0, delayUp=0, carryTrim=true, offset=0, flightModes=0, curveValue=0, curveType=0})
	model.insertMix(8,2, {name="GainNr", switch=21, source=100, weight=25, multiplex=0, speedDown=0, speedUp=0, delayDown=0, delayUp=0, carryTrim=true, offset=0, flightModes=0, curveValue=0, curveType=0})
	`
function "clearmix" deletes all rows from 0 - 63
 the (run) function only runs one times

But if I run the script twise, there will be a fourth line. And I think this is not okay
Please try my script and have a look to CH9 after running the script twise.
	
https://github.com/EdgeTX/edgetx/files/7870189/RK-StabiRx_2.3.xx.zip

ParkerEde, 2022-01-15:
I found out, model.deleteMix only deletes one line of a Channel in one cycle of the run() function.
So I have to run() 64 times to make sure that all rows of a channel are removed and only on the 65th run I insert the mixer rows once. This now always works reliably, but I think it is only a workaround. The real error is apparently in the model.deleteMix function.
Anyway, I leave it now so :-)

pfeerick, 2022-01-16:
Except `model.deleteMix()` is only supposed to delete one mixer line at time - so you _should_ need to run it 64 times in order to delete 64 mixes (or at least run it for `model.getMixesCount(channel)`) if you want to clear any pre-existing mixes on a channel? Unfortunately `model.deleteMixes()` removes all the mixes, so isn't usable here. 

If there is sufficient need or interest in some form of `model.deleteMixCh()` function being added, that might happen in the future.

ParkerEde, 2022-01-17:
perfect. Thank you