EdgeTX/edgetx #1405: Companion should display commit number of compiled binary.

piotrva, 2022-01-10:
**Describe the bug**
In my opinion, in order to make issue reporting easier Companion should include commit number of the build in "About" screen.
This is how firmware version is presented in radios, where in addition to defined version you can see also commit number.

Also can fix year to 2022 ;)

**Screenshots**
[image]

**General information**
 - Companion: 2.7.0 "dev", Jan 10 2022, #1402

Eldenroot, 2022-01-10:
Yes, also a github report link should be added

pfeerick, 2022-01-11:
Thanks for this... I was just thinking that the other day :)

PR #1415 (2022-01-12): Include commit Number in About Companion page

Fixes #1405 

Summary of changes:

Define GIT_STR in version.h.in as in radio FW (stamp.h.in)
display git Commit number in mainwindow.cpp (About Companion Page) including a link to the commit on github 
display a link to open a new request/ issue
closed 2022-01-12 by pfeerick