EdgeTX/edgetx #1407: Moving scroll wheel closes throttle warning

piotrva, 2022-01-10:
**Describe the bug**
When in warning screen (throttle warning) the warning is closed also by rotating a dial.

**To Reproduce**
Trigger throttle warning, turn dial/knob. Warning is closed

**Expected behavior**
The warning should not be closed, as turning the knob by accident is highly possible. Also this is not the case for switch warning, which need clicking something actually.

**General information**
 - EdgeTX version: 2.7.0 nightly https://github.com/EdgeTX/edgetx/commit/5defeeedf49d9803e3a39bae4012bd9eb4b6c199
 - Hardware: RM TX12

pfeerick, 2022-01-11:
Oh, nasty... it should require the roller be pressed as well - unless the throttle is lowered, that is.

eshifri, 2022-01-11:
I cannot reproduce it on TX16S - works as it should: requires a press on a roller.
BTW, the message on the screen is "Press ANY key" - only RTN and ENTER work.

piotrva, 2022-01-11:
@eshifri - this is only B&W radios related as colorlcd targets handle warnings in different way.

pfeerick, 2022-01-11:
I just saw the associated PR - code is currently looking for `!getEvent()` so B&W handling is probably different - and @piotrva basically just confirmed that as I was writing.

MervDale, 2022-01-11:
Also touching anywhere on the screen on the TX16S will cancel the throttle warning

PR #1408 (2022-01-11): Rotary encoder movement not to close throttle warning
Fixes #1407

Summary of changes:
Checking only key presses for closing throttle warning instead of any event.
This replicates the behavior for switch warning.
closed 2022-01-11 by pfeerick