EdgeTX/edgetx #1411: LUA mixer script refresh rate >=50ms on TX16S ?

lombartz, 2022-01-11:
Even on an empty model with one simple mixer script which only measures the difference between the previous and the actual getTime() value the output ist stuck at 5 ticks wich means the refresh rate is 50ms / 20Hz.

It should be "around 30ms" what is already slow, but 50ms is barely usable for any realtime controls.

Any chance to improve that ?

Tested on TX16S hall with EdgeTX build from actual sources.

jfrickmann, 2022-01-12:
NEVER use Lua for real time controls!! 

Lua runs in the GUI thread that also does other things, e.g. battery check, power button etc. and it has no guarantees whatsoever for QoS. And so many things incl. other scripts can hose your Lua script with fatal consequences if you rely on it for control of your model. That is also stated in the documentation.

And since a round trip of the GUI thread takes _at least_ 50 ms, that's also what your script will take. At least!

lombartz, 2022-01-13:
Thanks for your answer, I am aware of the risks and that there always has to be a underlying fallback mixer.
It's a pity that there seems to be no other way than LUA to calculate the actual stick speed for instance.