EdgeTX/edgetx #1412: reduce risk of corrupt radio.yml

gagarinlg, 2022-01-11:
The radio.yml is written relatively often. When the write does not finish the configuration may be broken, depending on the amount of data laready written, in ways hard to detect. 
Instead of truncating the file at every write I propose this procedure:
1. open radio_new.yml (or another meaningful name
2. complete writing to the new file, including a checksum
3. delete old file, rename new file, renaming is (normally an atomic operation on file systems)

for reading:
1. try to open regular config file (radio.yml)
2. if it exists, check the checksum
3. if it dows not exist or checksum missmatch, try to open the "radio_new.yml" file
4. if it exists, check the checksum
5. if the checksum of eitehr file is good, just run normally
6. if no checksum is OK warn the user about a broken configuration and revert to defaults

I think some of the problems we currently see sometime are coming from partially written config files.

piotrva, 2022-01-11:
ad. 6. reverting to defaults is not a good option, as sometimes people may edit files manually (as with yaml files it is possible). Also broken file does not actually means it should be completely replaced by defaults - sometimes you might restore quite a bit of it.
I think a checksum warning should be displayed and then option to:
- force load
- force load & fix checksum (aka for developers)
- cancel load (do nothing...)
- reset model to defaults

gagarinlg, 2022-01-11:
I was talking only about radio.yaml, for now, what renders some of your options pointless. The idea of having a popup with some options is good though. I do not know if this is needed for models, Maybe the otehrs know how often those are written.
When anly talking about radio.yaml:

> * force load & fix checksum (aka for developers)

this is the same as "force load", as a new write will be triggered

> * cancel load (do nothing...)

which is essentially "go back to defaults", as there are not configurations read in at this point. A way to connect to USB and edit the file would be good here.

> reset model to defaults

this just does not apply to radio.yml

piotrva, 2022-01-11:
Oh I see, I did not read that it is related to radio.yaml
So these two options you mentioned would be ok, also maybe with some warning with force load: "your radio might not work properly".

gagarinlg, 2022-01-11:
I think so.

The broken file should be copied to another name, so that it can ba analysed later. I'll update the description for that

rotorman, 2022-01-11:
One could add an additional YAML element to the very top, e.g. "ManuallyEdited" that if set to yes, will skip checksum verification on first read. The radio would write it automagically back with correct checksum and change ManuallyEdited to no. This way one can still have checksum, but easy user modifcation, w/o having to fix the checksum value manually or via some script every time.

gagarinlg, 2022-01-11:
I added this to the description

stephenaa, 2022-01-18:
I think the checksum should be placed in the top of the file. That way we can reliably differentiate between the file not having a checksum (old file or manually edited) and a corrupt file (the start of the file is assumed to be written but the remainder not)

gagarinlg, 2022-01-18:
Checksum in the beginning is harder to implement.

stephenaa, 2022-01-18:
Agree - but I think putting it in the end will have us miss the exact case wr try to mitigate. 

Sendt fra min iPhone

piotrva, 2022-01-18:
> Checksum in the beginning is harder to implement.

Not at all - if we calculate checksum of field values only (not whole TXT file) then it is just about order of fields in file.

To support this - today I had an issue, where (somehow) part of my channels were set to speedDown: 128
[model01.yml.txt](https://github.com/EdgeTX/edgetx/files/7891513/model01.yml.txt)

I do not know why (this happened only to one model) but it scared me a lot, when I was not able to disarm my quad!

stephenaa, 2022-01-18:
I intend to do some code for this tomorrow. I think I have an approach that will work. 

Sendt fra min iPhone

stephenaa, 2022-01-20:
I have some code running now. It adds a CRC16 in front of the file, and can detect CRC errors on load. Also a "manuallyEdited" field is checked (and cleared) when loading.

To keep it simple I suggest we do not make a UI for this for now, but instead let the radio:

- Copy broken config to new name (for later inspection)
- If a "open radio_new.yml" exists read it (this is created on write and copied to radio.yml as last step)
- If not: Revert to default
-  Show an alert what was done to resolve the broken configuration file (One of "Backup config loaded", "Reverted to defaults")

I think this allows devs. to get to interesting data as well as making it as painless as possible for users when the situation occurs.

gagarinlg, 2022-01-20:
Sounds good, but I would prefer that you delete the old and rename the new file, as a rename is atomic for the filesystem.

gagarinlg, 2022-01-20:
Sounds right to me.

rotorman, 2022-01-20:
Just tiny addition: in case of checksum error and `manuallyEdited` set to true, trigger a save.

stephenaa, 2022-01-22:
Ahh, fun detail. The test fails because the radio.yml has no checkup (and no manuallyEdited) - The radio.yml used for test of course needs to have manuallyEdited set in order or it to be usable with the new code without raising an alert.

Where does the test pick up radio.yml ?

pfeerick, 2022-01-23:
I'm pretty sure it unzips this for the TX16s and using the bin => https://github.com/EdgeTX/edgetx/blob/main/radio/src/tests/model_25_tx16s.otx

It then looks like the test then blew up on the conversion from 2.5 => 2.6

PR #1481 (2022-08-31): reduce risk of corrupt radio.yml #1412
Fixes #1412 

Summary of changes:
Adds CRC16 checksum to radio.yml when saved.
Adds save to temporary file, folowed by rename to radio.yml
Adds check on load, with fall-back to temporary file
closed 2022-08-31 by pfeerick