EdgeTX/edgetx #1414: File-->Download is greyed out in companion.

rc-charlie, 2022-01-11:
**Describe the bug**
I just installed EdgeTX Companion and a post installation pop up suggests downloading the latest firmware by using File---> Download. However, after setting up my radio profile, requiring bluetooth (for head tracker), LUA and noheli, When I go to the file drop down menu, the Download button is greyed out and I'm unable to download a suitable firmware build.

**To Reproduce**
Download EdgeTX Companion, Set up radio profile, file dropdown menu, download button.

**Expected behavior**
I was expecting a firmware build to include bluetooth, lua and noheli to be downloaded for me to flash to the radio.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX Companion 2.5.0
 - Radiomaster TX16S

Nerenye, 2022-01-11:
EdgeTx does not have build servers yet. If you need a specialised build you need to compile it for your self or ask somebody to do it

prebuild files you can get from here:
https://github.com/EdgeTX/edgetx/releases

or via Flasher:
https://github.com/EdgeTX/flasher/releases/tag/latest

or Buddy:
https://buddy.edgetx.org/#/flash?version=v2.5.0

rc-charlie, 2022-01-11:
> 
Thank you. Is there a place I can ask a kind someone to compile one for me, or find out how I can do it? many thanks.

Nerenye, 2022-01-11:
discord would be a good place for that
https://discord.gg/SsWrDdW3

rc-charlie, 2022-01-11:
> discord would be a good place for that https://discord.gg/SsWrDdW3

Thank you, I shall close this now :-)

closed 2022-01-11 by rc-charlie