EdgeTX/edgetx #1418: x9d+se get stuck when sdcard plug in.

gcrisis, 2022-01-12:
I don't know wether the problem is due to edgex. But the problem happens after edgetx is installed,so I hope you can help me.
When I plug in sdcard then I connect my controller to PC through usb. After disconnect PC then get into TOOLS page,the problem happen.If I reboot ,the problem occur at the boot menu.

https://user-images.githubusercontent.com/13515569/149076247-4f5220a6-776d-4422-bd02-65c7b804c703.mp4

hope get your answer as soon as possible,Thank you!

pfeerick, 2022-01-22:
Is this EdgeTX 2.5? If not, can you try [EdgeTX 2.6.0-RC3](https://github.com/EdgeTX/edgetx/releases/tag/v2.6.0-rc3)? I *think* your radio may be stuck in a bootloop, and not clearing the screen between cycles.

gcrisis, 2022-01-23:
> 

> Is this EdgeTX 2.5? If not, can you try [EdgeTX 2.6.0-RC3](https://github.com/EdgeTX/edgetx/releases/tag/v2.6.0-rc3)? I _think_ your radio may be stuck in a bootloop, and not clearing the screen between cycles.

yes,EdgeTX 2.5.Thank you for your reply.I will have a try to  2.6.0.

pfeerick, 2022-01-24:
2.6.0 will most likely be released shortly, so if you don't want to do a double install, might be good to wait a day or so for that ;)

gcrisis, 2022-01-25:
the problem solved after i update to 2.6.0.but I find some other little problems.my radio is frsky x9d plus se 2019.
1.after i update firmware 2.6.0 through https://buddy.edgetx.org/#/flash?version=v2.6.0,my radio cannot power up when i long press it,nothing happened.the power led keep blue for ever.I unplug my battery and reconnet then ok.
2.i set power up delay to 1s,but the radio costs about 2-3s,power down delay is OK;
hope for your reply,thank you!

pfeerick, 2022-01-26:
1. Not sure what happened there - after dfu flash, you may have had to disconnect the USB lead, and then it should have powered up normally. But it sometime shappens. 

2. Did you flash the X9D+2019 firmware? As I can't repo this on my X9D+2019... no delay and 1s delay remove or speed up the power-on delay animation for me.

gcrisis, 2022-01-26:
Thank you pfeerick.
i make a video .It's not a big problem to me,just give you a feedback.
https://user-images.githubusercontent.com/13515569/151108161-48779f97-df28-44af-ac49-923a4b93d7bc.mp4

pfeerick, 2022-01-26:
Well, that's just annoying, All I can suggest is it is probably something different about the SD cards we're using... as the X9D+2019 should basically be the same as the SE just the SE has all the extras fitted. I will close this issue though now, as the underlying problems has been resolved with the upgrade, and the poweron delay will be tracked here... https://github.com/EdgeTX/edgetx/issues/1503

closed 2022-01-26 by pfeerick

gcrisis, 2022-01-27:
ok