EdgeTX/edgetx #1422: Building EdgeTX

K2AAE, 2022-01-12:
I setup a build in WSLq / Ubuntu 20.04 and built a release but it is 2.4 where is a guide on getting the current build? I couldn't find anything about adding features to EdgeTX anywhere.

piotrva, 2022-01-13:
As you check out main branch from git it should get you a built of 2.7 - please check what commit have you checked out from git.
And I suppose discord server would be better place to get such support ;)

nostromo123, 2022-01-31:
cmake -DPCB=X10 -DPCBREV=EXPRESS -DHELI=NO -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Release ../

That is the command line, where you add or remove features.

Building for my X10 Express with HELI removed and LUA MIXER added.