EdgeTX/edgetx #1427: BF conf Lua stops works on 2.6 RC2

RipperGOT, 2022-01-13:
The Betaflight lua config stopped working on Edge TX 2.6 RC2
Once the lua has been executed, it remains on the initial screen with the message: Initialising... 
Not tested in 2.6 RC1, in 2.5 works. 

Radio: TX16s

Another question. 
I have also observed that the touch screen does not work in the shutdown confirmation with the model still powered. You have to press the wheel to turn it off.
Open a new issue with this? 

[image]

pfeerick, 2022-01-14:
Without looking this any further, I wonder if the culprit is the same as the one for https://github.com/EdgeTX/edgetx/issues/1346. I do see one use of `getFieldInfo` in the BF lua repo but I don't know if it is related. I'll try to pin it down on the weekend as I can reproduce this.

jfrickmann, 2022-01-16:
Are there any sensors with a name that could be read as a number, e.g. "0007"?

klutvott123, 2022-01-16:
```getFieldinfo``` could be the culprit as in 1.5.0 it's used to get the value of the "stateSensor" we're using to detect a connection. For 1.6.0 we're using ```getRSSI()``` to detect the connection instead. The latest nightly may work for you https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20220113.5522. It's backwards compatible with older BF versions so if you're not on 4.3 that's not a problem.

jfrickmann, 2022-01-16:
Please try #1423 and see if that helps

RipperGOT, 2022-01-16:
No, not work! But i tried the last nightlies release of BF Conf Lua and it work with EdgeTX 2.6 RC3.

May be BF problem. 

https://github.com/betaflight/betaflight-tx-lua-scripts-nightlies/releases/tag/v20220113.5522

pfeerick, 2022-01-16:
I'm not sure it is the culprit - have also just tried the 1.5.0 BF Conf lua with TX16S running 2.6.0-RC3 and it had no issue loading the `Betaflight setup` and is reading values from the FC. So I think there is something else going on, but it seems like the changes to 1.6.0 Lua for BF 4.3.0 have resolved it anyway.

JoeriCapens, 2022-01-16:
I've also done some testing. To me it seems the EdgeTX version and #1423 fix don't matter. But the Betaflight 1.5.0 LUA scripts only work for Betaflight 4.2 , for Betaflight 4.3 you need the nightly release.

@RipperGOT any chance that in addition to your EdgeTX 2.5 to 2.6-RC2 upgrade you also upgraded to Betaflight 4.3 ?

klutvott123, 2022-01-16:
@JoeriCapens that's true, there has been changes to the msp over telemetry protocol which means that 1.5.0 won't work with 4.3. BF lua 1.6.0 will still work with 4.3 and older

pfeerick, 2022-01-16:
@JoeriCapens Ah, that makes sense... as I thought I did get the betaflight configurator refuse to load at one point recently. But, that was with a quad that is running 4.3-RC2 as it has onboard ELRS... Hopefully something similar happened for RipperGOT since the nightly is working.

RipperGOT, 2022-01-17:
> I've also done some testing. To me it seems the EdgeTX version and #1423 fix don't matter. But the Betaflight 1.5.0 LUA scripts only work for Betaflight 4.2 , for Betaflight 4.3 you need the nightly [...]

Effectively...
I was testing with a HM Crux3 Nlr, if I remember correctly when I bought it, it had a custom 4.3. Now if it is with an official 4.3 RC2. That's the problem! 

@pfeerick Same quad? 😉

pfeerick, 2022-01-17:
Yup, that's the same bugger ;)

pfeerick, 2022-01-17:
@RipperGOT Yup, that's the same one! 😆 Yeah, it came with a custom betaflight - think it was actually 4.2.11 with the ELRS on top, but I can't be sure... I *was* running a pre-release of the 10.8 betaflight configurator so that I could download 4.2.11 in the first place so that could have been why it worked also (I'm running the RC of that now also)... either way, it's on 4.3RC2 now also, so I'll need to bump my Lua configurator so that i can work with it as well.

closed 2022-01-17 by pfeerick

le-rena, 2022-01-17:
my tbs agent lite wont loadin my  jumper t18. says( ?:0:attempt to call and nill value)

le-rena, 2022-01-17:
this is the error message
[image]

pfeerick, 2022-01-17:
@le-rena Try the firmware from this build - the fix for this was merged yesterday : https://github.com/EdgeTX/edgetx/suites/4956514608/artifacts/144056740

le-rena, 2022-01-19:
still does the same thing im gonna start all over again tonight
thanks for the help