EdgeTX/edgetx #1428: Touch screen does not work in the shutdown confirmation with the model still powered

RipperGOT, 2022-01-13:
Only works the wheel for confirmation.

**To Reproduce**
-Turn on TX 
-Turn on RX 
-Turn off TX and try to press the touch screen for confirmate the shut down.

 - EdgeTX 2.6 RC2
 - TX16S

pfeerick, 2022-01-13:
Is this related? It seems the LCD being off when the poweroff triggered is the issue there :  https://github.com/EdgeTX/edgetx/issues/1335

RipperGOT, 2022-01-14:
> Is this related? It seems the LCD being off when the poweroff triggered is the issue there : #1335

Omg! I did a good search... Sorry!!!!🤦‍♂️

Yes it's the same but backwards.  
If I have the screen on and try to turn it off, the touch screen does not work.
With the screen off:  turn off the radio, the screen does not appear until you move a stick, the screen turns on and the touch works. 

Close it if you consider that it is duplicated.

pfeerick, 2022-01-14:
Duplicate, but also different... since I thought touch still worked if screen was on, so it's probably not what I thought it was causing it...

gagarinlg, 2022-01-17:
I'll take a look

PR #1469 (2022-01-21): force backlight on when displaying full screen dialogs
Fixes #1428

Summary of changes:
when displaying a full screen dialog on color lcd radios, the backlight is forced on and therefore the touch screen is enbaled, if the radio has one.
commit ea29a68f (2022-01-21): force backlight on when displaying full screen dialogs
fixes #1428
closed 2022-01-21 by raphaelcoeffic