EdgeTX/edgetx #1429: Newly insert SRFF LS should be off instead of on

wimalopaan, 2022-01-14:
A newly inserted SRFF LS (without set and reset switch) is in state ON. I think this should be OFF.

pfeerick, 2022-01-14:
You mean Sticky LS? As yes, that is a little misleading, so we'll have to see why OTX has always evaluated it to active on both colorlcd and B&W even though clearly the set condition has been been given, let alone triggered yet. :-/

wimalopaan, 2022-01-14:
Created PR #1430 to fix that.
But maybe there are reasons for not changing this ...???

pfeerick, 2022-01-14:
Possibly, but if you've not set the V1 (set) condition yet I fail to understand in what situation the Sticky should be true when newly created, since it should only be true/on once V1 is true. 🤦

wimalopaan, 2022-01-14:
If there is no v1, then the condition always yields to a set of te LS