EdgeTX/edgetx #1435: Linux Companion Audio

pfeerick, 2022-01-15:
2.6.0-RC3

It seems like the Linux AppImage Companion (but a simulated model does play audio) is having problems when trying to play/preview audio tracks... i.e. pressing the blue play button here ...

[image]

... results in ...

```
(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-WARNING **: 21:19:48.582: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so: undefined symbol: gst_query_parse_bitrate

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-WARNING **: 21:19:48.583: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so: undefined symbol: gst_query_parse_bitrate

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GLib-GObject-CRITICAL **: 21:19:48.583: g_object_set: assertion 'G_IS_OBJECT (object)' failed

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-CRITICAL **: 21:19:48.583: gst_object_ref: assertion 'object != NULL' failed

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-CRITICAL **: 21:19:48.583: gst_bin_add_many: assertion 'GST_IS_ELEMENT (element_1)' failed

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-CRITICAL **: 21:19:48.583: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-CRITICAL **: 21:19:48.583: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-CRITICAL **: 21:19:48.583: gst_ghost_pad_new: assertion 'GST_IS_PAD (target)' failed

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-CRITICAL **: 21:19:48.583: gst_element_add_pad: assertion 'GST_IS_PAD (pad)' failed

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-CRITICAL **: 21:19:48.583: gst_object_unref: assertion 'object != NULL' failed

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-WARNING **: 21:19:48.586: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstopengl.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstopengl.so: undefined symbol: gst_video_aggregator_get_type

(EdgeTx_Companion_2.6.0-fac0eed-x86_64.AppImage:1874): GStreamer-WARNING **: 21:19:48.588: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstopengl.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstopengl.so: undefined symbol: gst_video_aggregator_get_type
```

pfeerick, 2022-01-15:
I wonder if after looking at https://github.com/intel/gstreamer-media-SDK/issues/178 and https://github.com/Stellarium/stellarium/actions/runs/1325544502/workflow if `libqt5multimedia5-plugins` is needed when building?

pfeerick, 2022-01-15:
Ok, it seems like it might be a mismatch in gstreamer versions - the version Companion in the AppImage was built against vs the version installed system.

elecpower, 2022-01-18:
> Ok, it seems like it might be a mismatch in gstreamer versions - the version Companion in the AppImage was built against vs the version installed system.

Deja vu...the issue that keeps on giving

BigChinoDon, 2022-06-20:
Funny the sounds play fine (2.7.1) when you simulate the model.

gagarinlg, 2023-03-18:
is this still a thing?

pfeerick, 2023-03-21:
Yes, no change. Sound is still working if you simulate the model, but not if you press the blue play button in Companion:

```
EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.059: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstplayback.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstplayback.so: undefined symbol: gst_type_mark_as_plugin_api

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.059: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so: undefined symbol: gst_type_find_helper_for_data_with_extension

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.059: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so: undefined symbol: gst_type_find_helper_for_data_with_extension

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GLib-GObject-CRITICAL **: 19:20:58.059: g_object_set: assertion 'G_IS_OBJECT (object)' failed

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-CRITICAL **: 19:20:58.059: gst_object_ref: assertion 'object != NULL' failed

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-CRITICAL **: 19:20:58.060: gst_bin_add_many: assertion 'GST_IS_ELEMENT (element_1)' failed

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-CRITICAL **: 19:20:58.060: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-CRITICAL **: 19:20:58.060: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-CRITICAL **: 19:20:58.060: gst_ghost_pad_new: assertion 'GST_IS_PAD (target)' failed

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-CRITICAL **: 19:20:58.060: gst_element_add_pad: assertion 'GST_IS_PAD (pad)' failed

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-CRITICAL **: 19:20:58.060: gst_object_unref: assertion 'object != NULL' failed

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.061: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so: undefined symbol: gst_video_meta_set_alignment

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.061: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so: undefined symbol: gst_video_meta_set_alignment

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.062: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so: undefined symbol: gst_video_meta_set_alignment

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.062: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstopengl.so': /lib/x86_64-linux-gnu/libgstgl-1.0.so.0: undefined symbol: gst_video_format_info_component

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.063: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so: undefined symbol: gst_video_meta_set_alignment

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.063: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so: undefined symbol: gst_video_meta_set_alignment

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.063: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so: undefined symbol: gst_video_meta_set_alignment

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.063: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so: undefined symbol: gst_video_meta_set_alignment

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.064: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstopengl.so': /lib/x86_64-linux-gnu/libgstgl-1.0.so.0: undefined symbol: gst_video_format_info_component

(EdgeTx_Companion_2.9.0-x86_64.AppImage:3867): GStreamer-WARNING **: 19:20:58.064: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so: undefined symbol: gst_video_meta_set_alignment
```

elecpower, 2023-03-21:
@pfeerick PR #3339 may fix this as it is using a different AppImage builder. I tested and it works for me.

elecpower, 2024-01-28:
@pfeerick is this still an issue with the 2.10 AppImage?

pfeerick, 2024-01-28:
Looks like it... get this error vomit at the console when hitting the play button for a Play Track SF (and doesn't revert back to play button  - stays as stop button) :cry: I do get audio when I run the simulator up though... 

```
(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.061: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstplayback.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstplayback.so: undefined symbol: gst_query_parse_selectable

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.061: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so: undefined symbol: gst_debug_log_id
[W] unknown():0 - Error: "" : "no element \"identity\""

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.062: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcoreelements.so: undefined symbol: gst_debug_log_id

(AppRun.wrapped:4745): GLib-GObject-CRITICAL **: 07:35:33.062: g_object_set: assertion 'G_IS_OBJECT (object)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.062: gst_object_ref: assertion 'object != NULL' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.062: gst_bin_add_many: assertion 'GST_IS_ELEMENT (element_1)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.062: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.062: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.062: gst_ghost_pad_new: assertion 'GST_IS_PAD (target)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.062: gst_element_add_pad: assertion 'GST_IS_PAD (pad)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.062: gst_object_unref: assertion 'object != NULL' failed

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.062: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so: undefined symbol: gst_navigation_event_new_touch_up

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.062: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so: undefined symbol: gst_navigation_event_new_touch_up

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.062: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so: undefined symbol: gst_navigation_event_new_touch_up

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.063: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstopengl.so': /tmp/.mount_EdgeTXYDcpyz/usr/bin/../lib/libudev.so.1: version `LIBUDEV_247' not found (required by /lib/x86_64-linux-gnu/libgudev-1.0.so.0)

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.063: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstkms.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstkms.so: undefined symbol: gst_video_content_light_level_from_caps

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.063: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so: undefined symbol: gst_navigation_event_new_touch_up

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.063: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstwaylandsink.so': /lib/x86_64-linux-gnu/libgstwayland-1.0.so.0: undefined symbol: gst_video_center_rect

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.063: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so: undefined symbol: gst_navigation_event_new_touch_up

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.064: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so: undefined symbol: gst_navigation_event_new_touch_up

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.064: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstxvimagesink.so: undefined symbol: gst_navigation_event_new_touch_up

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.064: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstopengl.so': /tmp/.mount_EdgeTXYDcpyz/usr/bin/../lib/libudev.so.1: version `LIBUDEV_247' not found (required by /lib/x86_64-linux-gnu/libgudev-1.0.so.0)

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.064: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstkms.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstkms.so: undefined symbol: gst_video_content_light_level_from_caps

(AppRun.wrapped:4745): GStreamer-WARNING **: 07:35:33.065: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so': /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstximagesink.so: undefined symbol: gst_navigation_event_new_touch_up

[... cut ...]
(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.065: gst_element_link_pads_full: assertion 'GST_IS_ELEMENT (src)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.065: gst_element_get_static_pad: assertion 'GST_IS_ELEMENT (element)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.065: gst_ghost_pad_new: assertion 'GST_IS_PAD (target)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.065: gst_element_add_pad: assertion 'GST_IS_PAD (pad)' failed

(AppRun.wrapped:4745): GStreamer-CRITICAL **: 07:35:33.065: gst_object_unref: assertion 'object != NULL' failed

```

elecpower, 2024-01-28:
Thanks for the update. Since you mention it do remember seeing the plug-in but not sure why I did not think to include. Seems I need to revisit.

pfeerick, 2024-01-29:
Made a little progress here... https://github.com/EdgeTX/edgetx/tree/cpn-gstreamer ... linuxdeploy gstreamer plugin is now being used... will require `patchelf` added to the edgetx build container as it's a dependency of the plugin... i'm now getting an actual error message from the GUI, and the following in the log, so just need to figure out what extra gstreamer plugins are needed now...  🤞 🙏 

```
/usr/lib/x86_64-linux-gnu/gio/modules/libgiognomeproxy.so: undefined symbol: g_uri_is_valid
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognomeproxy.so
/usr/lib/x86_64-linux-gnu/gio/modules/libgiolibproxy.so: undefined symbol: g_uri_is_valid
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiolibproxy.so
/usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so: undefined symbol: g_source_set_static_name
Failed to load module: /usr/lib/x86_64-linux-gnu/gio/modules/libgiognutls.so
[W] unknown():0 - GStreamer; Unable to pause - "qrc:/sounds/0.wav"
[W] unknown():0 - Error: "Your GStreamer installation is missing a plug-in."
[W] unknown():0 - Warning: "No decoder available for type 'audio/x-wav'."
```

nostromo123, 2024-02-29:
Was surprised, that this issue is still there. Remember, when I build Companion from the source in Edge 2.6 and 2.7, I did not have that issue (play track) here under different Debian releases. 
I used the ubuntu building documentation and converted it to Debian. But my old Debian notes don't work anymore, 

At the moment I am on Edge 2.9 and using the app.image.

Just curious, what is the situation now with Debian 12 (bookworm).  To do so, I start with 

https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04

_Setting up the build environment for EdgeTX_

looks ok!

_Building EdgeTX firmware for the radio_

also ok!

_Building Companion, Simulator and radio simulator libraries_

but both stop with ERROR!

So I'll open an issue on that. Hope, there is still some interest in that. ;-)

br KH

pfeerick, 2025-11-01:
Non-reproducible on Linux Mint 22 with current main... I think we can call this done! :)

closed 2025-11-01 by pfeerick