EdgeTX/edgetx #1436: Crossfire doesn't connect with onebit enabled and model's receiver number different from 00

Lanzo98, 2022-01-15:
**Describe the bug**
Crossfire doesn't connect with onebit enabled and model's receiver number different from 00. Both telemetry and connection is lost 

**To Reproduce**
Steps to reproduce the behavior:
Enable one bit mode
Change model's receiver number, different from 00

**Expected behavior**
Telemetry and connection between Rx and TX works

**General information**
 - EdgeTX version 2.5.0
 - Hardware Frsky X9D

raphaelcoeffic, 2022-01-15:
Is this the good old bug with the Receivers that used to ignore the receiver number but don't now? Have you tried to rebind with a receiver number set?

Lanzo98, 2022-01-15:
Honestly I don't even know the meaning of the receiver number, haven't tried to rebind, I simply putted 00

pfeerick, 2022-01-16:
Receiver number is so that you can make it so that only the right model setup will control a given model. i.e. instead of all your receivers operating regardless of the model you have selected in the transmitter, receivers will only work if you have the same id number assigned on both sides. Or if it is 0, it is supposed to be ignored. Maybe.

Lanzo98, 2022-01-16:
Yes, the strange thing is that on Open tx the model was saved and binded with receiver number 02, but when I putted the models on Open tx it didn't connect to the Rx. But with 00 works

gagarinlg, 2023-03-18:
is still still an issue?

closed 2023-05-05 by pfeerick