EdgeTX/edgetx #1439: Logical switch Edge not working correctly

borsti2000, 2022-01-15:
Bug description:
New Model in 2.6.0 RC3
Creation of logical switches L01-L09 --> Edge, V1 SG down, V2 0,0 / imeadedly,  L01 -1 sec L02 -2 sec L03 -3 sec .....
Input in Mixers 1-9 (CH1 l01 / CH2 L02....)

Expected:  Switch SG down, all chanel otput to 100% and switch off after 1 to 9 sec

Result : No changes on CH1. 

Additional Test: change of L01 to 5 sec. It seems that L01 is input of CH2.

eshifri, 2022-01-15:
HW details, please?

borsti2000, 2022-01-15:
Oh sorry. TX16s. You need more info?

borsti2000, 2022-01-15:
I still try to test hats happens.
Now I have some additional info.
I seems that for example the Log.switch #L05 use the parameter of the line from L04.
To test it, I create two Lines (L05 + L04) with Edge and the same parameters. Exept the time (L05 --> 1 sec / L04 --> 5 sec)
In the chanal I use only the L05 (which should have 1 sec) but in simulation the output chanal is 5 sec "on"

eshifri, 2022-01-15:
Interesting: looks like I cannot reproduce it on TX16S. At least the LS works - did not try to manipulate channels.

pfeerick, 2022-01-16:
I've not tried this on hardware yet, but simulator is screwy with this config (which I "think" is what @borsti2000 is describing?). The config file reads fine, just simu is misbehaving on the mixes page (logical switches all seem fine). 

[image]

[issue1439.etx.zip](https://github.com/EdgeTX/edgetx/files/7876528/issue1439.etx.zip)

commit cdd933d6 (2022-01-16): Fix shift for logical switches in raw sources

Resolves #1439
PR #1449 (2022-01-18): Various Companion fixes
- [x] Fixes #1446
- [x] Fixes #1447 
- [x] Fixes #1419 
- [x] Fixes #1439 

Summary of changes:
- Fix (de-)serializing enable param for backlight custom function
- Fix OFF backlight
- Fix power ON/OFF delay
- Fix flight mode trim disabled
- Fix trim switch index
- Fix deserializing SLIDER[1,2]
- Fix shift for logical switches in raw sources
commit 929bb396 (2022-01-18): Fix shift for logical switches in raw sources

Resolves #1439
closed 2022-01-18 by pfeerick