EdgeTX/edgetx #1452: Jumper T-Pro and companion 2.3.14, cannot read models and settings from radio, error: 'Can't extract RADIO/radio.bin'

unites, 2022-01-16:
*Before submitting,  search the open issue if there is already a report of the bug*
Original Issue Report under OpenTX here... https://github.com/opentx/opentx/issues/8830

Worth noting I don't see a Jumper T in the Firmware library so maybe that is in the works?

Quote...
```
Dear All,

I'm new to OpenTX. Just bought a Jumper T-PRO and did a few things, ran into this problem right away. I've been trying to find a solution for almost two days and haven't found it yet, ...did I miss something, should do or download a more appropriate version? Thanks in advance!

Background,

Download and prepare the sdcard file sdcard-128x64-2.3V0038.zip on the micro SD Card and insert it into the radio. Now radio can talk.
Download OpenTX Companion, install, and added profile settings as Jumper T-Lite, the rest are default
No problem with PC and radio connection. The SD Card radio is read by Explorer complete with its folders and there is another drive with some bin files.
Everything is still original
...haven't flashed anything to radio, just insert SD card, create and name a model, set it as FUTABA (no further setup).
Steps to reproduce the behavior:

Power up the radio via bootloader
Connect USB C cable
Run Companion
Goto Read/Write>Read Models and Settings From Radio
Expected behavior:
The process should be seamless to read and copy models and radio settings

Screenshots:
Like other 'Can't extract RADIO/radio.bin' screenshots

General information:

OpenTX version... (how can i check firmware version of Jumper T-PRO?)
Hardware brand new Jumper T-PRO, brand new SD Card microSDHC 16GB
OpenTX Companion 2.3.14 on Windows 7
```

piotrva, 2022-01-16:
Please, do not repost your problem/issue, as you opened also #1453 which is on the same issue.
Also I may recommend going with such questions to edgeTX Discord server: https://github.com/EdgeTX/edgetx.github.io/wiki/Community-Guidlines

unites, 2022-01-16:
> Please, do not repost your problem/issue, as you opened also #1453 which is on the same issue. Also I may recommend going with such questions to edgeTX Discord server: https://github.com/EdgeTX/edgetx.github.io/wiki/Community-Guidlines

From my point of view these were entirely separate issues, not a "repost".  One having to do with manually trying to acquire the firmware from this repo due to an error on my part.  The other attempting to use the edgetx software to load SD Card data. 

In the process of troubleshooting this I have learned that these are related in so far as your development of software related to the Jumper T Pro was occurring on a development branch that has not been released.   It was at that point you could draw a correlation between the two. 

Apologies here, I got my first Transmitter and drone today in the mail.  This is my first time looking at your software.  The manufacturer

closed 2022-01-16 by unites

piotrva, 2022-01-17:
That's no problem!
From my POV it was connected as a single issue: where can I find firmware for my radio.
Good to hear that you managed to unbrick your Jumper and hope you enjoy edgeTX.
Feel free to ask for further help on Discord.
Have a nice day!