EdgeTX/edgetx #1453: EdgeTX Jumper T Pro Firmware

unites, 2022-01-16:
I am new to this so not sure if this is all on the manufacturer or the community.  I accidently flashed the Jumper T Lite Firmware onto my T Pro, naturally none of the controls work 🔰.  I can't seem to find firmware anywhere for it to fix this problem. Neither OpenTX nor EdgeTX has a version available in their repos, the manufacturer also does not have a version on their website.

Would be nice if there was a EdgeTX version of the firmware so I could use my controller again, ha ha.  I think it's pretty much bricked until I either get the OpenTX firmware from the manufacturer (Jumper) or the communities in either OpenTX or EdgeTX write one.
🥺

piotrva, 2022-01-16:
It is supported by: https://github.com/EdgeTX/edgetx/releases/tag/v2.6.0-rc3
Just go to Assets and get edgetx-firmware-v2.6.0-rc3.zip and you have your firmware in this archive.
If you cannot enter bootloader on your radio: https://github.com/EdgeTX/edgetx/wiki/Unbrick-your-radio

piotrva, 2022-01-16:
Also: get from the same source matching Companion release (it will not work with older companion, nor OpenTX companion)

unites, 2022-01-16:
> Also: get from the same source matching Companion release (it will not work with older companion, nor OpenTX companion)

Thanks, looking into it now.  Forgot to check for release or other branches.  For the SD data is there an accompanying 2.6 release, or should I use the current latest release SD card data?

piotrva, 2022-01-16:
This should do for you: https://github.com/EdgeTX/edgetx-sdcard/releases

unites, 2022-01-16:
Awesome this works.  End result here is that It appears like the transmitter I purchased is pretty new and everything related to it has not been officially released, for OpenTX and EdgeTX.

closed 2022-01-16 by unites