EdgeTX/edgetx #1455: Unused code for colorlcd Model Notes/Checklist display

piotrva, 2022-01-16:
**Describe the bug**
There seem to be unused code for colorlcd:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/gui/colorlcd/view_text.cpp#L57

As READ_FILE_BY_LINE is hard-coded to 0 in:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/gui/colorlcd/view_text.h#L29

If this is not prepared for future development I suggest it to be removed, as it makes any improvements made in this file to be done in two versions.
Otherwise, if it is used somewhere - a comment on it would be appreciated. 

**General information**
 - EdgeTX version: https://github.com/EdgeTX/edgetx/commit/8150b00823ca0a37ade72c41bb2fa35cbe31e594
 - Hardware: colorlcd

pfeerick, 2022-01-16:
It's card coded at view_text.h#L29, BUT is also used to either define code when on or off

When on:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/gui/colorlcd/view_text.h#L56

When off:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/gui/colorlcd/view_text.h#L101

@eshifri What say you? 😁 
[image]

eshifri, 2022-01-16:
I think it is OK to remove it. It was left as an older and reliably working implementation in case there would be problems with new one. Since the current version is working, we do not need old code any more.

piotrva, 2022-01-16:
Yup, those differences in .h file are also looking like unused code, as still - this define is always 0. Also I have not spotted it in any built directives (despite fact that it should be defined using #ifndef for such case).

So IMO it looks like this define might be deleted together with code for case it is 1.

piotrva, 2022-01-16:
@eshifri 
Ok, so tomorrow I will delete it and make PR.

PR #1456 (2022-01-21): remove READ_FILE_BY_LINE from colorlcd
Fixes #1455

Summary of changes:
- Remove old, unused code and a define to enable it from view_text implementation for colorlcd
closed 2022-01-21 by pfeerick