EdgeTX/edgetx #1458: Rebind Flysky rx

Mikey1968, 2022-01-17:
I been using the flysky rxs in myy planes for weeks using version 2.5 , but when I updated to 2.6 RC2 , I have to rebind my rxs everytime I fly . Not sure if this is a bug or what

pfeerick, 2022-01-17:
D8 or D16? As it seems there may be a bug affecting both OpenTX and EdgeTX for D8 receivers as I've seen a few mentions now of this, and that turning the internal module on and off again seems to fix it (no rebind necessary). 

What Transmitter? And you're on RC2, not RC3?

Mikey1968, 2022-01-17:
Jumper T18 pro2 
There was no RC3 at the time I used RC2
I went back to 2.5 until there's a fix

pfeerick, 2022-01-22:
I've just been using a FS-iA6B receiver with effectively 2.6.0-RC3 and had no issues with needing to rebind. Changing models and power cycling the radio. TX16S+MPM.

kasatka60, 2022-01-31:
EdgeTX 2.6, Jumper T-lite multimodule, flysky resiver. You always have to bind the receiver when you change the model

kasatka60, 2022-01-31:
Can this bug be fixed?

Mikey1968, 2022-02-01:
That's not the problem. I did that in 2.5 but when you upgrade to 2.6 then you have to rebind every time. I went back to 2.5 and works fine

kasatka60, 2022-02-01:
Tomorrow / the day after tomorrow I will check again at my place.

pfeerick, 2022-02-02:
Can you be more specific about the receiver than just "Flysky receiver"? Anyone have any idea as to what model of receiver your are using? i.e. I have been using genuine Flysky FS-iA6B and FS-iA10B receivers and haven't needed to rebind, but in my case it has been with TX16S, haven't tried the T-Lite yet.

kasatka60, 2022-02-02:
jumper t-lite edgetx 2.6 multimodule 1.3.3.7, Flysky X6B, FlySky FS-iA10B, FlySky FS-iA6B

kasatka60, 2022-02-03:
Everything works fine for me.

tommp100, 2022-05-26:
I am having a similar experience with 2 Flysky receivers (FS-IA6B) and a Radiomaster TX16S.  Only one will retain its binding (the one last bound).  If I attempt to bind the second receiver it will retain its binding but the first one will lose its binding.  I've assigned discrete numbers to the receivers (field next to 'bind') with no effect. I have not attempted using a different version of EdgeTX

tommp100, 2022-05-26:
f

kasatka60, 2022-06-27:
Change the receiver number in the settings and everything will work.

tommp100, 2022-06-27:
Changing the receiver number worked.
Thanks for the prodding.  I had given up on ever using my second Flysky FS-IA6B with the TX16S.
Just for anyone reading this:
The first thing I did was power up the model that was bound.
When I changed the receiver number I got the "telemetry lost" announcement.
So I put it  back to the original receiver number and got the "telemetry recovered" announcement.
I put in the bind plug on the unbound Flysky receiver.
I changed the receiver number as you suggested.
Powered up the receiver and bound it as usual.
I tested both receivers and the telemetry to each seems to be rock solid and independent.
Thank you.

pfeerick, 2022-11-26:
Glad you got it working, good to know for myself also. I knew on the upcoming EL18 with AFHDS3 rebinding with the same RX number was a problem (as there are settings stored in each slot), but as I didn't realise this would clobber on the MPM with AFHDS2A.

closed 2022-11-26 by pfeerick