EdgeTX/edgetx #1459: Edge TX Companion 2.6.

Wingy1690, 2022-01-17:
When I open a airplane file in Companion 2.6 I get the following message. 
[image]
[image]
I did not get this in 2.5
Any Ideas?? wingy1690@gmail.com

pfeerick, 2022-01-17:
Can you attach one of the offending files? (change the file extension to `.zip` so github will accept it). And is this with 2.6.0-RC3?

pfeerick, 2022-01-22:
@Wingy1690 Really need those files if you want any resolution to this ;)

ParkerEde, 2022-01-26:
I got the same error message. 
Horus X10 EXPRESS: 
If i convert the otx bin files (SD Card as datasource) with the standalone simulator, copy the created yml to the etx archiv and open it in companion. Only D16 is affected.

pds59, 2022-01-27:
radiomaster t16 s same problem with companion 2.6 , if this can help in my case, solved by loading old  b. up. otx file

pfeerick, 2022-08-12:
I'm going to assume since there was no further commentary or files uploaded this issue is gone. Please open a new issue and also provide a reproudcable test case - i.e. a zip of the radio/models folders from the radio, or the OTX/ETX file that triggers this, with `.zip` as the extension so github accepts it.

closed 2022-08-12 by pfeerick