EdgeTX/edgetx #1464: V 2.6 On-screen Keyboard and Character list

chucky-whiskers, 2022-01-17:
The keyboard, as well as the character list selected by the scroll wheel on my TX16S Max are missing some very valuable characters such as punctuation or various symbols such as degrees°; percent%; comma,; period.; parentheses(); and ampersand&. I frequently used some of these in OpenTX and was disappointed to find they are not available in EdgeTX yet.

raphaelcoeffic, 2022-01-19:
Just so you know: we are migrating to a better system, including a better character set support, so that we should be able to have some more in 2.7.

gagarinlg, 2022-06-11:
You might want to check the next nightly, but it is not flight ready yet.

chucky-whiskers, 2022-06-11:
Thanks for reaching out!

gagarinlg, 2023-02-26:
close, because we are now using the LVGL keyboard, which has the missing characters

closed 2023-02-26 by gagarinlg