EdgeTX/edgetx #1465: Global Functions too restricted

chucky-whiskers, 2022-01-17:
I have been forced to create special functions on each individual model that are the same for a throttle cut function. In OpenTX I was able to program a throttle cut as a global function overriding the value of a channel (channel 3 for mode 2) when a switch was in a certain position, as well as another global function to play a track for "throttle cut". I would like all special function abilities to be available in global functions on my TX16S Max.

koaxheli, 2022-01-17:
Did not exist in OTX - you are wrong.

chucky-whiskers, 2022-01-17:
I had to look through my old models to see. I guess I just had the "play track" on a global function. I stand by the idea of enabling more functions in the global functions menu, though. I apologize for my mistake.

gagarinlg, 2023-03-18:
With model templates, I think this is solved. Any differing opinions?

JimB40, 2023-03-19:
I think it solved. Having different kind of aircrafts user would have to switch off thotlle cut every time quad is selected.

gagarinlg, 2025-04-12:
won't fix

closed 2025-04-12 by gagarinlg