EdgeTX/edgetx #1470: Add "RF Channel" to options when XN297 protocol is selected for Multi-module

Zebble, 2022-01-18:
The "RF Channel" option seems to be missing when using the XN297 Dump protocol.  This allows picking the channel on the module you specifically want to dump.  Otherwise it will always default to channel 0 if you don't select the AUTO subprotocol.

raphaelcoeffic, 2022-01-18:
@Zebble Could you please detail which radio you are using? On the color lcd radios, we take the options directly from the MPM, so that it should not happen. On B&W, it is kind of similar, but using the older protocol.

Zebble, 2022-01-18:
I'm using a RadioMaster TX16S, running EdgeTX 2.5.0 and an iRangeX+ external multimodule running firmware 1.3.3.0 (Debug version for Serial debug to work for XN297 dumping).  I'm using the external module in this case, as it allows debugging.

raphaelcoeffic, 2022-01-23:
I could not find much on the XN297 dumping. Can you please explain a little more how that works? My MPM firmware does not seem to have it.

Zebble, 2022-01-23:
I believe the XN297 protocol will only appear if you install the debug version of the MPM firmware.  I'm using the "usbdebug" firmware on my iRange4+ so I can use the USB port to see the output of the XN297 dump.

When selecting the XN297 there is apparently supposed to be an "RF Channel" option to pick the specific channel you want to sniff.

The XN297 dumping feature itself allows sniffing of the air to detect any XN297-formatted packets.  It will then decode these into something the dev team can use to add a protocol for a currently unsupported model.

https://www.multi-module.org/advanced-topics/serial-debugging

https://www.multi-module.org/advanced-topics/protocol-dumping

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Advanced_XN297Ldump.md

That last link details what the configuration fields do in OpenTX wrt this protocol.

Zebble, 2022-01-23:
Just did a quick code comparison, and I think changing this line in multiprotocols.cpp

 {MODULE_SUBTYPE_MULTI_XN297DUMP,  4, false,      STR_SUBTYPE_XN297DUMP, nullptr},

To this:

{MODULE_SUBTYPE_MULTI_XN297DUMP,  5, false,      STR_SUBTYPE_XN297DUMP, STR_MULTI_RFCHAN},

Add adding to the defines:

#define STR_MULTI_RFCHAN                     QT_TRANSLATE_NOOP("Multiprotocols", "RF Channel Number")

gui_common.cpp looks like it will need a respective change as well:

{MODULE_SUBTYPE_MULTI_XN297DUMP,  5, false, false,  STR_SUBTYPE_XN297DUMP, STR_MULTI_RFCHAN},

Will fix the issue.  

I can do a pull request if that's easier?

pfeerick, 2022-01-24:
Please do!

gagarinlg, 2023-05-05:
any news on this one?

richardclli, 2023-05-06:
Should be related to
https://github.com/EdgeTX/edgetx/issues/2713

Let me test this out.

mha1, 2023-05-06:
The missing rf channel field is already fixed in PR #3435  but the protocol is excluded from selection at all because MPM is not listing this protocol with proto scan using regular firmware. Learning this protocol is offered by MPM using debug fw I enabled this protocol with the correct parameters. Please re-test with PR check firmware in #3435.

richardclli, 2023-05-07:
> The missing rf protocol field is fixed in PR #3435 but I excluded XN297 from selection at all because MPM is not listing this protocol with a protocol scan using regular firmware. Learning this protocol is offered by MPM using debug fw I enabled this protocol with the correct parameters.
> 
> If the MPM debug firmware includes XN297 in the protocol scan results it'll show this:
> 
> [image] [image]
> 
> Please re-test with MPM debug firmware and PR check firmware in #3435.

Yes, your fix solved this as well.  I tested and all goes well.  Changing the RF Protocol options works properly.

PR #3504 (2023-05-23): fix: MPM protocol options missing in various protocols
Fixes #2713, fixes #3046 (partly #3500), and fixes #1470

Summary of changes:
- use getMultiOptionTitle() instead of rfproto->getOptionStr() as getMultiOptionTitle() takes updated MPM status in account
- changes to mpm_settings.cpp to check for MPM changing subtype for DSM auto
closed 2023-05-22 by pfeerick