EdgeTX/edgetx #1472: Trim6 up/down missing in logical switches after conversion from otx/bin

ParkerEde, 2022-01-19:
EdgeTX 2.6.0RC3
1. Open a OTX File in Companion with a model that has a logical switch OR Trim6u Trim6d. So far it's okay.
[image]

2. Save the file in etx file format
3. open the new etx file in companion, check the logical switches => Trim6 up is missing
[image]

4. start companion simulator of this model, check the logical switches => Trim6 down is missing
[image]

Same with the standalone Simulator after converting models.

Please see attached otx and etx file
[EdgeTX_Trim6_bug.zip](https://github.com/EdgeTX/edgetx/files/7896099/EdgeTX_Trim6_bug.zip)

ParkerEde, 2022-01-19:
Trim5 up/down is okay in Companion after conversion:
[image]

But the simulator shows this:
[image]

There seems to be a shift of the switches here.

pfeerick, 2022-01-19:
I believe this was fixed in #1449, which will either be part of RC4 or the final 2.6.0 release. 

If you are on Windows you can download the installer for the latest 2.6.0 branch Companion here => https://github.com/EdgeTX/edgetx/suites/4969517456/artifacts/144868852

ParkerEde, 2022-01-19:
thank you. This companion works correctly. Sorry for the double post.

closed 2022-01-19 by ParkerEde

ParkerEde, 2022-01-20:
Hi @pfeerick . To prevent double posts, do you think it is possible, that the companion will also be find in ASSETS of the nightlies?
Kind regards

pfeerick, 2022-01-20:
Probably not, as it takes too long to build. I'll look if there is some way to link to already built versions though.

If people are running beta versions then there is *some* expectation they use the latest version GitHub actions has built if there is an issue and also look at commit logs before opening issues (or at least tried). And if that happens and an issue still gets opened, so be it.