EdgeTX/edgetx #1474: Fields are filled incorrectly in the companion

koaxheli, 2022-01-19:
Transmitter TX16S 
cpn V2.6.0rc3

In the companion the sources in the mixers are
are not read in correctly or not at all.

[image]

 - EdgeTX version (e.g. 2.6.0 rc3)
 - Hardware (e.g. RM TX16S)

pfeerick, 2022-01-19:
Can you check if the build of Companion mentioned here (sorry, on phone and can't link it directly) has this problem still?
 https://github.com/EdgeTX/edgetx/issues/1472#issuecomment-1016389617

ParkerEde, 2022-01-20:
Hi @pfeerick, yes I will check....

ParkerEde, 2022-01-20:
I verified otx / etx file. It looks good to me:
[image]

Is this test okay so far, or should I test anything else in this case, or in another way?

ParkerEde, 2022-01-20:
hi again,
If I use LS for sources in mixes (as you wrote) then L64 is missing after saving in etx format and reopen:
[image]

commit 83f03a81 (2022-01-21): Fixed logical switch index check in raw sources

Resolves #1474
commit c0c91ae8 (2022-01-21): Fixed logical switch index check in raw sources

Resolves #1474
closed 2022-01-21 by raphaelcoeffic