EdgeTX/edgetx #1476: Change desktop background with an error. Nv14

Ktr128, 2022-01-22:
If a non-default theme is used, to change the desktop background, the file must be placed not in the folder with the active theme, but in the folder with the default theme.

To Reproduce 
Choose any other theme than the default theme. Put the desktop background file in the default theme folder, reboot nv14. the desktop background will change.

Expected behavior
Place the desired desktop background in the current (active) theme.

General information
EdgeTX versionnightly-dcbdde2b 2022.01.17
Hardware nv14 (flysky nirvana)

pfeerick, 2022-01-22:
I *think* it was supposed to be the other way... if default background as present, it would be used, otherwise theme provided background was used. And the default background path needs to be changed from `THEMES\EdgeTX` to `IMAGES`, so that there is no confusion as to it being realated to the theme. 

But at the same time, more work is needed on the theme configuration stuff, as IMO it would be nice to be configurable per model.

Ktr128, 2022-01-22:
I agree with you. But it will be enough that the desktop background file is taken from the active theme folder.
In any case, taking the desktop file only from the default theme folder, even when another theme is active, is not true...

pfeerick, 2022-01-23:
Can I just check what name you are giving the file? As themes can have two different backgrounds - one for 480x272 screens like the TX16S, and 320x480 ones for the NV14... so the filename is a little different for the background for themes vs the default background. 

> background_480x272.png (a background image for your theme in 480 x 272 pixel resolution, e.g. for TX16S, T16, X10, X12S)
> background_320x480.png (a background image for your theme in 320 x 480 pixel resolution, e.g. for NV14)
> https://github.com/EdgeTX/themes#description-of-an-edgetx-theme

Ktr128, 2022-01-24:
> background_320x480.png
I was wrong. When creating a theme, I named the file after the default theme - background.png
Now I checked it by renaming it to background_320x480.png.
Everything is ok, it works well.
Sorry for reporting the false bug.

closed 2022-01-24 by Ktr128

pfeerick, 2022-01-24:
No problem... something that needs documenting better somehow ;)