EdgeTX/edgetx #1477: The variometer does not work. Nv14

Ktr128, 2022-01-22:
When the model is connected to the internal radio module, there is no vertical speed in the list of sensors. Height and pressure are correct.
But without vertical speed, the variometer does not work.

To Reproduce
Connect flysky FS-CAT01 height sensor to native receiver.
Then connect the receiver to nv14.
Go to the list of sensors.
The vertical speed sensor will not be listed. There is also no way to manually create a vertical speed sensor.

Expected behavior
The vertical speed sensor should appear in the list of sensors. Or there should be a formula in the calculated sensors section, which will allow you to make the sensor manually.

**Screenshots**
[image]
[image]

General information

EdgeTX versionnightly-dcbdde2b 2022.01.17
Hardware nv14 (flysky nirvana)
                 altitude sensor  flysky FS-CAT01

raphaelcoeffic, 2022-01-22:
@Ktr128 As far as I know, the `FS-CAT01` is a simple pressure sensor, which sends only that value. I know some have implemented some kind of vario feature for it, but this has been done by using the first derivate of the altitude, which is clearly too slow and sub-optimal.

Real varios do implement the vertical speed directly in the plane, and sending the vertical speed to the radio. Please also note that `FS-CAT01` is sold explicitly as an Altitude sensor, nothing more. And in fact, it does not even calculate the altitude itself, it just sends the air pressure back to the radio, which is supposed to calculate the altitude out of it.

closed 2022-01-22 by raphaelcoeffic

pfeerick, 2022-01-22:
Sadly as Raphael said, while it is an altitude/pressure sensor just like the Frsky sensors, it isn't a high precision one, i.e with 0.1m "high precision" accuracy so isn't suited for use as a variometer, but can be used as one... With around 1m precision. You can see for yourself if you move it up and down what the resolution of the sensor is. What it is good for is pressure, temperature and approximate height, and a crude variometer. What we really need is a FS-CAT02 with high precision sensor...

raphaelcoeffic, 2022-01-22:
> What we really need is a FS-CAT02 with high precision sensor...

That is a point we should raise with FlySky ;-)

raphaelcoeffic, 2022-01-22:
> Do you mean that flysky should develop a new sensor with vario functionality?

Yeah, that's what @pfeerick meant. 

> After all, he is in great demand for gliders!

Agreed, I could use one as well! We will discuss this with FlySky.

Ktr128, 2022-01-22:
I will be very glad if flysky will listen to us.
This is also in their interest if they want to grow and compete with other manufacturers.

eshifri, 2022-01-23:
Actually, FS-CAT01 measures only pressure and temperature. Altitude is calculated by EdgeTX. It is possible to add calcu;ation of the vertical speed as well. After all, there is no better solution.
See PR #1485

Ktr128, 2022-01-23:
Eshifri, thank you very much! You are a magician!!!