EdgeTX/edgetx #1478: SA-SD is not work, while functional switch 1-4 is None

r1000ru, 2022-01-22:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
While not setted not "None" type of Functional switches 1-4, switches SA-SD is not active.

**To Reproduce**
Set type "None" for function switch 1 disables FS1 and SA-switch
Set type "None" for function switch 2 disables FS2 and SB-switch
Set type "None" for function switch 2 disables FS3 and SC-switch
Set type "None" for function switch 2 disables FS4 and SD-switch

**Expected behavior**
No dependency between S(A-n) and Functional switches.

**General information**
 - EdgeTX 2.6-RC3
 - Hardware Jumper T-Pro

pfeerick, 2022-01-22:
@3djc I think this is relevant to you guys also as I was able to repo this on the PR8834 build I had on hand.

3djc, 2022-01-22:
Fix : https://github.com/opentx/opentx/commit/65fc03f4b22d52eaf93c77a15bcc61259c9b9f48

closed 2022-01-23 by pfeerick