EdgeTX/edgetx #1482: Not possible to flash iXJT module on 2.6.0RC3/Nightly

mhon44, 2022-01-22:
Hello, 

I have run into a problem with flashing Frsky firmware to radio

**Describe the bug**
Since updating to RC3, even latest nightly, not possible to flash internal module on Frsky X-Lite.
Reverted FW back to 2.5 - all went good.

**To Reproduce**
Update to 2.6.0RC3 on radio with iXJT module.
Flash internal module.

**Expected behavior**
Update of the internal iXJT ACCST module to version 2.x 

**General information**
 - EdgeTX version 2.6.0RC3 / Nightly (e.g. 2.4.0 or nightly and #commit number)
 - Hardware FrSky X-Lite

raphaelcoeffic, 2022-01-23:
@mhon44 where is it exactly failing to flash? Can you please add a screenshot?

raphaelcoeffic, 2022-01-23:
By the way, have you tried to flash a receiver via S.PORT? It is basically the same code being used at both places. The only difference is which power is line is turned on. It is the exact same S.PORT line used to flash a internal module, external module or a receiver via S.PORT output.

mhon44, 2022-01-23:
> @mhon44 where is it exactly failing to flash? Can you please add a screenshot?

Right at the beginning - FW sync error not responding. Unfortunately don't have screenshot

Receiver via S.Port worked without any issues.

Nerenye, 2022-01-25:
I tried with horus x12s and it works for me

closed 2023-09-05 by raphaelcoeffic